qmk/qmk_firmware #3701: Space Cadet Shifts sends extraneous input with KC_GESC

edward-shen, 2018-08-20:
When typing with normal alphanumeric characters a quick shift + character sends the correct output. However, when space cadet shifts are enabled, Using either shifts for a `Shift + GESC` to access ~ outputs `~)` and `~(` when the keycombo is pressed quickly enough.

For reference, I'm using a DZ60 with pretty vanilla firmware (no special macros or anything).

Let me know if there's any additional information needed, but I think this should be reproducible in all keyboards.

edward-shen, 2018-08-22:
Is there a possible workaround for this issue?

drashna, 2018-08-22:
Not that I'm aware of.

edward-shen, 2020-01-16:
It was mentioned that #5226 fixes this issue, but #5226 was replaced by #5277—@XScorpion2, can this issue by closed?

XScorpion2, 2020-01-16:
Ya, this issue should be closed. I just tested it to make sure this doesn't occur with the latest code too and it all checks out.

noroadsleft, 2020-01-16:
Thanks for reporting this!

closed 2020-01-16 by noroadsleft