qmk/qmk_firmware #3718: Compiling  ymdk_np21

fragilityV1, 2018-08-21:
I'm trying to compile some custom firmware for my YMDK numpad.  Before compiling my edited keymap I wanted to try the default map first.  Based on the QMK documentation for the numpad, https://github.com/qmk/qmk_firmware/tree/master/keyboards/ymdk_np21, the command to compile the default keymap is:

make ymdk_np21:default:program

However, when I use the command I get the error message:

/usr/bin/env: ‘python’: No such file or directory
make[1]: *** [tmk_core/avr.mk:125: program] Error 127

What am I missing?  I've got Python installed, is something not in the correct directory.  I don't even have a env directory in bin.  Sorry if this is an obvious answer, this is my first time using QMK with this numpad.

fragilityV1, 2018-08-22:
Thanks for your help.  I'm making some progress.  I forgot to mention I'm using msys on Win10 for the compiling.  Based on the ECHO from msys I assumed it wasn't reading the Windows environmental variables.  So I added the Python27 directory to the msys PATH (after adding a bunch of garbage learning how it worked).  So now msys looks like it can find Python, but now I'm getting the error message:

Traceback (most recent call last):
  File "./util/atmega32a_program.py", line 22, in <module>
    import usb
ImportError: No module named usb
make[1]: *** [tmk_core/avr.mk:125: program] Error 1
Make finished with errors
make: *** [Makefile:544: ymdk_np21:default:program] Error 1

So atmega32a_program.py is looking to import a "usb" module, but it doesn't appear to be installed.  Is there something else I need to install?

fauxpark, 2018-08-22:
I think you may need to install pyusb with `pip install pyusb`. This information is in the [README](https://github.com/qmk/qmk_firmware/blob/master/keyboards/ymd75/README.md) for the YMD75, but not the NP21. I recommend reading that instead, it looks like you can just download HIDBootFlash and flash the .hex file directly instead of using this rickety Python script.

fragilityV1, 2018-08-22:
Thanks!  I'll give it shot tonight and let you know.  Thanks for the help!

fragilityV1, 2018-10-22:
I gave up trying to use QMK on the keypad.  I ended up going with another editor that was compatible with the existing firmware, but I can't recall for the life of me what I ended up doing.

drashna, 2018-10-22:
Ah, I'm sorry to hear that.

And I'm guessing BMC or TMK? 

And I'll close this, in that case.

closed 2018-10-22 by drashna