qmk/qmk_firmware #3730: Custom definitions in rules.mk?

sklarsa, 2018-08-23:
I'm working in `layouts/community/ergodox/sklarsa/`, and my keymap compiles and loads fine, but it appears that I can't use any newly defined settings from my rules.mk in my keymap.c file.

As a test, I set `MOUSEKEY_ENABLE = no` to confirm that rules.mk is being processed by the Makefile, which it is (the resulting binary is smaller than before).

But, if I add the following to my rules.mk:
```
TESTRULE = yes
```
and add the below snippet to my keymap (which should cause a syntax error), everything appears to compile correctly.
```
#ifdef TESTRULE
This is bad syntax
#endif
```

If I use an existing setting, like `MOUSEKEY_ENABLE`, set it to `yes` in rules.mk, and try to compile, the code block is processed by the compiler and I get a syntax error (which is expected)

Am I missing something? How can I add custom definitions to my rules.mk?

drashna, 2018-08-23:
That is absolutely doable, but the way you're doing it is "wrong".  Eg, it's not how the makefile stuff processes it. 

I use this a good deal in my userspace, so I can customize what features are enabled, per keyboard. 

https://github.com/qmk/qmk_firmware/blob/master/users/drashna/rules.mk#L14-L16

Eg, you want something like this: 
```makefile
ifeq ($(strip $(NO_SECRETS)), yes)
    OPT_DEFS += -DNO_SECRETS
endif
```

This also defines the "NO_SECRETS" setting so that my keymap.c file can use stuff like `#ifdef NO_SECRETS` too.

sklarsa, 2018-08-23:
> drashna/rules.mk#L14-L16

Perfect, got it to work. Thanks!

closed 2018-08-23 by sklarsa