qmk/qmk_firmware #3732: [Infinity Ergodox] Right keypad dying randomly

jessejanderson, 2018-08-23:
Since switching to qmk for my Ergodox (I'd been using the native firmware up until a few weeks ago), I've had an issue where my right keypad will just randomly stop responding to input. My setup has my right keypad plugging into my left keypad, and the left keypad plugs into the computer.

Unplugging and replugging it in fixes the issue.

I tried various ways of plugging in the keyboard, thinking at first that it was an issue with a lack of power. But I had the same issue whether I plugged directly into my Macbook, through the monitor, or through a powered USB hub.

It seems to happen most often when I am activating a layer with a hold-down modifier key. When it dies this way the lcd display shows that it is still in that layer (even though I'm no longer holding down the modifier key) but will not accept any input. Sometimes though, it will happen even when not using any layers or modifier keys (though I suppose it is possible that it could be related to a native modifier, e.g. shift, ctrl, cmd, opt).

These are the keymap files that I'm currently using: https://github.com/jessejanderson/qmk_firmware/tree/master/keyboards/ergodox_infinity/keymaps/jergo

jackhumbert, 2020-03-17:
@jessejanderson is this still an issue? It looks like that PR may have solved it.

jessejanderson, 2020-04-01:
@jackhumbert I'd test if I could, but I actually sold my Infinity Ergodox a year ago and got an Ergodox EZ.

zvecr, 2020-10-20:
Closing for now as the scenario cannot be retested, we can reopen if its still an issue.

closed 2020-10-20 by zvecr

CobaltCause, 2022-02-23:
So I actually have been having this exact problem for the past year or so and just been dealing with it, but recently I updated my firmware to no longer be several years old, and the interconnect stopped working entirely. I looked though the schematics and tested the relevant connections with a multimeter and it all seemed fine. I desoldered one half to disassemble it in case there was anything going on on that side of the board, which there wasn't, so I reassembled it. While putting it back together, I took some compressed air to the USB-A and USB-C ports and both halves of the keyboard, and now it works fine :facepalm: 

TL;DR: If you have this problem, try taking some compressed air to all of your keyboard's USB ports