qmk/qmk_firmware #3755: HHKB NKRO not enabled on startup

chadgrossman, 2018-08-26:
Hello,

I have 2 HHKB's with Hasu's controller (one BT one standard) and both seem to not like having NKRO enabled on plugin/startup (this is over USB on both, not BT). I've added "#define FORCE_NKRO" in config.h and "NKRO_ENABLE = yes" in rules.mk but both don't enable NKRO. If I toggle NKRO using MAGIC_KEY it does work. Is there any way to get this to enable on startup that I'm missing? I know recently this was an issue in TMK and Hasu was able to force NKRO while on USB but I can't seem to find the change he made in TMK and try to replicate here.

drashna, 2018-08-26:
The commit in question is this: https://github.com/tmk/tmk_keyboard/commit/f2761c9565e76706853a4a0b148c8e93369e49e5

chadgrossman, 2018-08-26:
Thanks for pointing that out, I got it working now by bringing over some of the changes from that commit.

closed 2018-08-26 by chadgrossman

drashna, 2018-08-26:
Awesome.  If you wanted to open a PR with the changes, that would be awesome!

chadgrossman, 2018-08-26:
I'll try cleaning it up a little and reducing the changes to be only in that rn42 folder and do that. Thanks again!