qmk/qmk_firmware #3764: QMK-DFU bootloader and :production target currently doesn't work for at90usb1286 MCUs

TerryMathews, 2018-08-26:
Was informed by @drashna on Discord that the QMK-DFU bootloader and :production target currently only works for the atmega32u4 target and not the at90usb1286.

I intend to work on adding support for the at90usb1286 MCU as I've got a vested interest in it working properly (TKC1800).

MxBlu, 2018-08-28:
Is there more to it than passing in value for FLASH_SIZE_KB, BOOT_SECTION_SIZE_KB and maybe F_CPU? It doesn't seem like the LUFA bootloader has any trouble compiling under those circumstances.

TerryMathews, 2018-08-28:
Just from trying to create a :production HEX and watching the console output, I know the underlying code to scrape MCU isn't working correctly. I haven't traced it step by step yet to see if it's right anywhere in the bootloader creation process yet.

My first step is going to be making sure the underlying bootloader code does compile correctly for at90usb1286 then attack the various build scripts.

MxBlu, 2018-08-31:
I was playing around the other day and got the bootloader to compile, though I am unable to test if it was functional.

```
MCU          = atmega32u4
ARCH         = AVR8
BOARD        = QMK
F_CPU        = 16000000
F_USB        = $(F_CPU)
OPTIMIZATION = s
TARGET       = BootloaderDFU
SRC          = $(TARGET).c Descriptors.c BootloaderAPI.c BootloaderAPITable.S $(LUFA_SRC_USB)
LUFA_PATH    = ../../LUFA
CC_FLAGS     = -DUSE_LUFA_CONFIG_HEADER -IConfig/ -DBOOT_START_ADDR=$(BOOT_START_OFFSET)
LD_FLAGS     = -Wl,--section-start=.text=$(BOOT_START_OFFSET) $(BOOT_API_LD_FLAGS)

# Flash size and bootloader section sizes of the target, in KB. These must
# match the target's total FLASH size and the bootloader size set in the
# device's fuses.
FLASH_SIZE_KB         = 32
BOOT_SECTION_SIZE_KB  = 4
```
We're looking at setting the values of MCU, F_CPU, FLASH_SIZE_KB and BOOT_SECTION_SIZE_KB based on the configuration, I'd try myself but I'm personally not too familiar with make scripts where you pass around variables, is the appropriate form using `export VAR` from the parent make script?

TerryMathews, 2018-09-03:
` [GCC]     : Compiling C file \BootloaderDFU.c"
In file included from ../../LUFA/../LUFA/Drivers/Board/LEDs.h:138:0,
                 from BootloaderDFU.h:55,
                 from BootloaderDFU.c:37:
../../LUFA/../LUFA/Drivers/Board/AVR8/QMK/LEDs.h:140:13: error: expected '=', ',
', ';', 'asm' or '__attribute__' before 'static'
             static inline void LEDs_Init(void)
             ^
BootloaderDFU.c: In function 'SetupHardware':
BootloaderDFU.c:244:2: warning: implicit declaration of function 'LEDs_Init' [-W
implicit-function-declaration]
  LEDs_Init();
  ^`

drashna, 2018-09-03:
There was a recent change to the build script. That may be why

MxBlu, 2018-09-03:
Uh oh, that was me... Check out Keyboard.h in lib/lufa/Bootloaders/DFU, it gets generated by the make script.

I resolved an issue where it was fetching the wrong value for product, and not appending " Bootloader" correctly under Windows, but it looks like despite my testing, it broke in another way...

Please check if there's an issue with Keyboard.h

TerryMathews, 2018-09-03:
https://github.com/qmk/qmk_firmware/pull/3786 should be reverted. I commented as needed.

The root of the issue here is that the makefile in lib/lufa/Bootloaders/DFU is hardcoded for a normal ATMega32U4. Trying to decide on the best way to carry the MCU declaration from the project's rules.mk into the DFU makefile.

drashna, 2018-09-04:
I'm not sure that the commit needs to be reverted.  I tested it on my system, and it compiles fine still (MSYS, Win10).

That said, make sure that you're on on avr-gcc 8.2 (or 8.1).  If you are, revert to 7.x, and see if that helps.

MxBlu, 2018-09-04:
https://github.com/qmk/qmk_firmware/pull/3848

I suspect the problem wasn't the version of avr-gcc, but rather grep itself. This fix _should_ work, but testing should be done before merging in.

As for the make scripting, I too am not too adept at it haha. I'll search things up and give my shot at it later.

TerryMathews, 2020-03-15:
@drashna wanted to touch this to update. I was recently reworking the TKC1800 codebase and decided to try the qmk_dfu bootloader again.

The build script detects the correct amount of space for the MCU, and :production does generate a bootloader hex that will flash and work on the TKC1800, but it won't jump back to program, whether I use the Esc key or dfu launch.

Same program hex flashed with the stock Atmel DFU jumps back with no issues.

I don't even know where to being troubleshooting this. I assume either the bootloader is occupying the wrong space, or the jump target address is wrong.

TerryMathews, 2020-03-18:
This is working now.

closed 2020-03-18 by TerryMathews