qmk/qmk_firmware #3769: GRAVE_ESC with GUI sends GUI+GRAVE, wanting just GRAVE

ar1a, 2018-08-27:
I was setting up my firmware and thought that was kind of strange. I have a 60%, and the desired behaviour was:

Esc = Escape
Shift+Esc = Tilde
Gui+Esc = Grave

But it sends Gui+Grave, which doesn't print the character. Is there a way to remedy this?

chadgrossman, 2018-08-27:
Do you have GRAVE_ESC_GUI_OVERRIDE defined? GRAVE_ESC works for me how you want unless I add that option which will then not print a character since it sends GUI+ESC.

kdb424, 2018-08-28:
If your keyboard has that enabled by default (my vitamins included had some bad defaults) You can simple fix it with 
#undef GRAVE_ESC_GUI_OVERRIDE

ar1a, 2018-08-28:
https://github.com/ar1a/qmk_firmware/blob/aria/keyboards/v60_type_r/config.h#L83 The only thing I have on is GRAVE_ESC_CTRL_OVERRIDE

leico, 2018-09-10:
If you can try my experimental pull request, first to prepare backup and recover way, secound try my pull request with set below.

`#define GESC_GR_IGNORED_MOD KC_LGUI`

or if you need both side GUI ignore

`#define GESC_GR_IGNORED_MOD_BIT ( MOD_BIT( KC_LGUI ) | MOD_BIT(KC_RGUI) )`

in your `keymap/config.h`

skullydazed, 2018-09-10:
I think this is working as designed. When you press GUI the GUI key will be registered, and when you press GESC it will see that GUI is pressed and register GRAVE instead of ESC. At no point is a key unregistered.

If you want it to work the way you describe you'll need to write some custom code. Our docs for that are here: https://docs.qmk.fm/#/custom_quantum_functions

closed 2019-11-21 by skullydazed

CharString, 2020-11-06:
> I hope to implement more easy customizable way in the future.

@leico Have your hopes come true or is one of the patches mentioned here still the fix?