qmk/qmk_firmware #3805: Extended space cadet shift

frederik-h, 2018-08-30:
My keyboard configuration includes an "extended space cadet shift":
tapping shift once, twice, thrice yields (, {, [.
I have found several references to similar configurations, for instance [this one](https://www.reddit.com/r/olkb/comments/5pof70/in_qmk_does_the_space_cadet_option_allow/).
It is not very difficult to set this up using a tap dance. But I think it could be worthwhile to include this function in QMK.

frederik-h, 2018-08-31:
The keycodes and the modifiers should also be configurable, see #3815.

Is there anyone else who would like to have such a feature in QMK? Would this be best implemented using tap dance or analogously to the current KC_LSPO?

sypl, 2018-09-02:
I'm going to guess you're a coder. Wouldn't this be annoying? If I want to enter }) then I'd have to: tap, tap, wait a bit, tap

frederik-h, 2018-09-03:
I have no long-term experience with it, but having used it for about a week, I must say I like it. Its a very convenient position since it involves almost no hand movement. You are right that it is slower than the usual way of typing sequences of parens and brackets, but it is not as bad as you might imagine, since the tapping has to be very fast and the wait time is correspondingly short. I still have two other ways to type those symbols on my keyboard ...

Lenbok, 2018-09-05:
@frederik-h Do you have a pointer (e.g. git commit) to your implementation? I'd like to give this a whirl.

Lenbok, 2018-09-05:
@frederik-h Thank you so much! I'll give that a go. I'll leave some questions over at the gist.

frederik-h, 2018-10-29:
I think now that I have used the functionality for a longer timer I can asses its usefulness better: I use one tap to produce ( a lot, two taps to produce { quite frequently, three taps to produce [ rarely, and four taps to produce < almost never. I am not sure that this feature should be included in QMK and hence I am closing this issue.

closed 2018-10-29 by frederik-h