qmk/qmk_firmware #3831: Feature Request: ASOFF_T(kc) ... Turns off Auto Shift when held key, when tapped kc sends

leico, 2018-09-02:
I am trying to create a keymap, it is featured auto shift.
I noticed that sometimes auto shift disturb any functions.
Titled feature is smart, useful solution to improve above issues I guess.

leico, 2018-09-14:
As far as I see source codes, mostly keycodes are already reseved.   If add more keycodes for this request, `SAFE_RANGE` are going to tight.   Can not use `...` to `..._MAX` method....

drashna, 2018-09-17:
This ... is complicated to add and would require addit it to the `tmk_core` and action code.  

So, while this should be possible, "complicated" is an accurate description.

jackhumbert, 2020-03-17:
I think for the time being, this would need to be implemented as a custom keycode with `process_record` - if you have any relevant code to share, feel free to drop it here!

closed 2020-03-17 by jackhumbert