qmk/qmk_firmware #3906: Suggestion: sharing user functions space such as user_library directory

leico, 2018-09-14:
Hello,

We can see that users creates useful features for your own in each `keymap.c` , especially LED performance.    I guess it is useful that there can simply includes in my `keymap.c`.    Perhaps some users want to share their created features. 
In order to realize it, how about to create a directory ?   Something like `user_library` . Below examples temporary named it.

For a this directory, QMK only provide

* no contains in this directory
    * users search features yourself, and put in it each
* `make` iterate only `user_library/*/rules.mk`
    * feature provider needs to write all complie sources in it

providers:

* needs to write `rules.mk`
* also to write `readme.md` and usage is kinder
* contains codes in each source files
     * like `process_record_fx`, `matrix_scan_fx`, etc...

users:

* need to find features yourself and put in `user_library`directory it each
* need to change `keymap.c`
    * like include `"fx.h"`, add provided keycode, add `process_record_fx` in `process_record_user` etc...

Now it is just an idea. Perhaps some issue occured to implement it, ex. `SAFE_RANGE`, conflicts other user function etc...
If you have any thoughts on that please share it.

jackhumbert, 2018-09-14:
Is there any reason this couldn't be done with the `users/` directory we have now? e.g. `users/jackhumbert/cool_rgb_thing`. @Wilba6582 and I were just discussing something similar for maintainers.

leico, 2018-09-14:
Sorry, It perhaps my over looking it. Check it later.

drashna, 2018-09-14:
Also, some of this is in the examples: 
https://docs.qmk.fm/#/feature_userspace?id=example

And I actually link to my "template" file in the docs for the functions: 
https://github.com/qmk/qmk_firmware/blob/master/users/drashna/template.c

Additionally, I've been trying to break my userspace up, more and more, so it's easier for users to take code that they want out of mine. 

Additionally, since I think I have the most prolific and "neat" userspace (I ***may*** be biased though), I tend to be the go to guy for questions about it, and help.   As well as a basis for. 

Though, I have been wanting to clean up the docs for the userspace stuff, and probably add additional examples that cover stuff like what you've mentioned.

leico, 2018-09-14:
@jackhumbert @drashna Thank you tell me references.  @drashna 's userspace is great.
yes, I concerned it is hard to handle all user, keymaps pull request.

drashna, 2018-09-14:
Welcome, and thanks!  

And for the most part, it's not hard to handle.  Time consuming, maybe!  But I handle most of the keyboard, keymap and userspace PRs, actually.    And if I see stuff that's a problem, or poorly implemented, I usually flag that and try to help.   

Also, if you're trying to implement something, you can always head over to our discord server, and myself or somebody can help out.   

Also, I tend to post code snippets when people ask "how do I do this".  And I'll sometimes add stuff to my code, just so I have it as a reference, later.

Also, if a userspace or keymap causes issues for a specific board, we to try to tag people to let them know that there is an issue, so they can fix it. 

But basically, we like helping out, and making sure things work. :)

leico, 2018-09-14:
Thanks a lot! Close this issue.

closed 2018-09-14 by leico