qmk/qmk_firmware #3916: RGB should play a sound to indicate the mode's speed

skullydazed, 2018-09-15:
Some of the RGB modes have very subtle animation changes between speeds, and it can be difficult to find the exact speed you want. The only real way to do so right now is to use QMK Toolbox to view debugging messages.

I'd like to propose a new define, `AUDIO_RGB`. This would be used in both RGBLIGHT and RGBMATRIX to play a sound every time the animation mode is changed. The pitch of this sound would correspond to the animation speed, with faster animations playing a higher note.

We can use this issue to both discuss the implementation details and to coordinate work on this feature if anyone would like to take it on. This is still a pretty rough idea, nothing is set in stone yet.

drashna, 2018-09-16:
Why not use audio clicky?  :P

That said, for the rgb underglow, .... some of the modes have 3 levels, some have 5 levels, and some ....

So this would be easier for the rgb matrix.

Also, why not handle it like the other audio events, and use defines for songs.

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc