qmk/qmk_firmware #3927: How to enable a new feature flag?

sypl, 2018-09-16:
I'm currently writing a new feature but am having trouble getting my build to pick up on the option I've created. I'm not sure what the secret sauce is to adding a new config option.

Here's a stripped down version of what I'm doing just to check if the option is on. For this example let's say the feature is called 'Superfly':

- in `quantum.c`, within `process_record_quantum`

```c
#ifdef SUPERFLY_ENABLE
  print("SUPERFLY_ENABLE");
#endif
```
- in my keymap `rules.mk`, located at `keyboards/sypl/keymaps/superfly_test/rules.mk`

```c
SUPERFLY_ENABLE = yes
```

I thought that would be it, and hitting any key would then send the string "SUPERFLY_ENABLE" to the console. However, it's not getting picked up it seems.

---

If I do something like this in the same `rules.mk`:

`KEY_LOCK_ENABLE = yes`

... and in `quantam.c`:

```c
#ifdef KEY_LOCK_ENABLE
  print("KEY_LOCK_ENABLE");
#endif
```

... then it prints "KEY_LOCK_ENABLE", as expected.

So I'm wondering whether there's a place I have to register SUPERFLY_ENABLE as a default? Or is threre something else I have to do to register my option with the firmware? I haven't found where options are registered or defaulted as yet. If anyone can point me in the right direction that would be fab.

Thanks!

MxBlu, 2018-09-16:
So what's actually setting the define isn't the "KEY_LOCK_ENABLE" make flag, but the makescript code in another makefile (common_features.mk has many examples) which would look like this in your case:

```
ifeq ($(strip $(SUPERFLY_ENABLE)), yes)
    OPT_DEFS += -DSUPERFLY_ENABLE
endif
```

OPT_DEFS feature an array of -D<define name> statements which are passed in to the compiler.

sypl, 2018-09-16:
Aha! I thought I'd looked everywhere, but clearly not. Thanks for that!

sypl, 2018-09-16:
@MxBlu Quick Q though: What is DSUPERFLY_ENABLE? It looks like some sort of annotation, but I can't find anything about it.

MxBlu, 2018-09-16:
The -D flag allows you to set a `#define` when compiling, see [here](https://www.rapidtables.com/code/linux/gcc/gcc-d.html) for an explanation. The lack of a space (`-D SUPERFLY_ENABLE` vs `-DSUPERFLY_ENABLE`) is a bit of a style thing, and also means the entire switch `-DSUPERFLY_ENABLE` is definitely going to be 1 array element in a makescript array, since spaces are separators for array elements.

sypl, 2018-09-16:
I see. Thanks very much for the explanation!

closed 2018-09-17 by sypl