qmk/qmk_firmware #3930: Ps/2 Mouse

LukMas, 2018-09-16:
Hi all! 

I'm trying to configure my Teensy2++ for using an external device connected via the PS/2 port. I've started with a generic mouse (that works for sure - I've tested it) before using the planned hardware, a trackball. Anyway, I'm not able to make it working. I've tried with the 2 best options, up to now (USART and Interrupt). I've just copied the code but there's no way to get it working. 
The keyboard is correctly programmed. I've built a matrix on the couple of breadboards with 4 rows and 4 columns and all the features that I've build are working. Just the mouse is missing. 

I've used the pins PD2 and PD5 for tx and clock for USART. 
For the interrupt, I've configured PD0 for clock and PF7 for data. 
Ground and voltage are connected to the correct pins of the board. 

If someone has got it working any help would be really appreciated. 

Luca

drashna, 2018-09-17:
Sounds like it's a communication issue then?   Not sure.

It may be worth looking at TMK/s PS2 to USB converter, as that may help you to get this working.

LukMas, 2018-09-18:
Thank you for the answer! 
What should I look, if I would like to check it by myself? I've not so much experience with MCU programming, but I know enough of C that I can git it a try, if I've a good hint.

LukMas, 2018-09-21:
I've solved it. I didn't know it, but a 5k drag is needed between the two lines CLK and DATA and the  line 5+. After reading carefully the doc I've noticed the "circuitry ... between module and controller". Then I've found one image where this was well explained and, after implementing it, it works.

closed 2018-09-21 by LukMas

drashna, 2018-09-21:
Awesome, glad to hear it!