qmk/qmk_firmware #3945: KC_EQUAL same as KC_LBRACKET

JonatannossreP, 2018-09-19:
For some reason KC_EQUAL does not send the right key but rather send the same keycode as KC_LBRACKET, please fix this.

noroadsleft, 2018-09-19:
Which keyboard is this for?

I just tested this and had no issues.

JonatannossreP, 2018-09-19:
All custom keymap, however I use all the standard files other then the layout. This is the only key that doesn't work. I have tried each key individually with numbers from 0-114 and know for sure that there are no issue with my matrix or layout otherwise and the almost complete keymap works well as well, except this single key.

JonatannossreP, 2018-09-19:
The keymap and all other files are here: https://www.dropbox.com/s/2ld2coci30pont9/kb.zip?dl=0

noroadsleft, 2018-09-19:
Handwired board?

Looks like you built your keymap from kbfirmware.com, which isn't an official QMK tool. Its code is nearly two years out of date and we have no control over it, though it is useful for custom handwired boards.

noroadsleft, 2018-09-19:
K, here's an updated .zip file.

The hex file should be good to flash. If you still have issues with KC_EQUAL and KC_LBRACKET outputting the same keycode, I suspect it's a hardware issue you're having.

The files I used to build the hex are in the `neoclavier_n17` folder.

[neoclavier_n17_jonatannossrep.zip](https://github.com/qmk/qmk_firmware/files/2398872/neoclavier_n17_jonatannossrep.zip)

JonatannossreP, 2018-09-20:
I installed the .hex, you compiled however it turns the keymap I gave you didn't use the KC_EQUAL but rather KC_KP_PLUS, which I used as a temporary work around. This means I do not know if the version you sent me would have worked. I now re-downloaded the whole qmk_firmware-master folder and deleted the old and used the files you provided to recompile the .hex, but again, the same issue. I find it hard to believe that it would be a hardware issue, in terms of soldering at least, since I have tested each key individually and with different KC_whatever and it is only the KC_EQUAL that does not work on any key, so I do not really know how I would fix it hardware wise.

JonatannossreP, 2018-09-20:
I found out the issue, was the Swedish input I use in Windows, which seems to read these input as the same thing and even display the same key when using a keyboard testing site. I where confused since this is the only keys which behave like this. Well it is nice when things have an easy solution in the end! Thanks for the help none the less.

closed 2018-09-20 by JonatannossreP

noroadsleft, 2018-09-20:
That's interesting that Swedish input "doubles up" on that key; I wasn't aware of that. I also never thought to consider the OS input language.

Glad you got it sorted.