qmk/qmk_firmware #3947: Update config.h template with new #defines

mechmerlin, 2018-09-19:
To save some more space in firmware, we can disable the handling for `action_get_macros` and `fn_actions`. They are depreciated so why not disable them? 

Add the following to `config.h` template 

```c
#define NO_ACTION_MACRO
#define NO_ACTION_FUNCTION
```

template files are located in: 

1. https://github.com/qmk/qmk_firmware/blob/master/quantum/template/avr/config.h
2. https://github.com/qmk/qmk_firmware/blob/master/quantum/template/ps2avrgb/config.h

drashna, 2018-09-19:
Better question is: 

Since these features are depreciated, do we want to actually disable them by default, and allow them to be turned on for compatibility?

noroadsleft, 2018-09-19:
@drashna 

> Since these features are depreciated

deprecated* :smile: 

@mechmerlin I'm okay with the spirit of these changes, but a lot of the keymaps in the repo have been ported from TMK, so we would have to go through each keymap that uses `fn_actions`/`action_get_macro` and refactor its functionality into `process_record_user`.

jackhumbert, 2020-03-18:
This was implemented in #7211.

closed 2020-03-18 by jackhumbert