qmk/qmk_firmware #3952: TMK Port: Ghost Squid

mechmerlin, 2018-09-20:
Reddit Post by GosuSan inquiring about the Ghost Squid was spotted: https://www.reddit.com/r/MechanicalKeyboards/comments/9heo2h/qmk_on_ghost_squid/

TMK repo: https://github.com/raphaelmor/tmk_keyboard/tree/master/keyboard/ghost_squid

A majority of bpiphany boards have already been ported to QMK and can be found in the keyboards/bpiphany directory. This is a good one to bring over.

tycm, 2019-03-25:
The port looks to be available here

https://github.com/BathroomEpiphanies/epiphanies_qmk_keyboard/tree/master/keyboards/ghost_squid_20140518

Not sure if it is missing a pull request or what, but it exists at least somewhat.

drashna, 2019-11-21:
@mechmerlin I smell ports.

fenuks, 2022-01-25:
I think this can be closed, support for ghost_squid has been merged into master.

closed 2022-01-30 by zvecr