qmk/qmk_firmware #3960: nix.shell incorrectly sets AVR include/lib paths for ARM

jmagee, 2018-09-22:
The `nix.shell` script populates the CFLAGS and ASFLAGS environment variables with the system include and library search paths for AVR:
```
  avr_incflags = [
    "-isystem ${avrlibc}/avr/include"
    "-B${avrlibc}/avr/lib/avr5"
    "-L${avrlibc}/avr/lib/avr5"
    "-B${avrlibc}/avr/lib/avr35"
    "-L${avrlibc}/avr/lib/avr35"
    "-B${avrlibc}/avr/lib/avr51"
    "-L${avrlibc}/avr/lib/avr51"
  ];
in
 ...snip...
  CFLAGS = lib.optional avr avr_incflags;
  ASFLAGS = lib.optional avr avr_incflags;

```

This causes problems when building for ARM because the AVR libc include files will be picked up instead of the ARM ones.  Amazingly, this does not seem to cause immediately obvious problems - mostly a few link time warnings, but otherwise it produces functional firmware for ARM-based boards.

Despite this surprising luck, it is not safe.  For example, while playing with the `AUDIO_CLICKY` feature, I found that (on an ARM board) rand() was returning values larger than RAND_MAX.  The problem was the AVR version of stdlib.h was picked up instead of the ARM one.

For a build of Planck rev6, simply removing the `avr_incflags` from shell.nix solved the problem.  However, it also caused any AVR build to fail to include the proper headers, so it is not a solution.

A workaround is to set `CFLAGS= ASFLAGS=` before running make with an ARM target.

I'm not sure on the best way to fix this properly.  On other systems, avr-gcc seems to know how to pick up the includes/libs without being told, but seemingly Nix needs some extra help.

andresilva, 2020-05-19:
Fixed in #8312?

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc