qmk/qmk_firmware #3975: Mouse movement on Ergodox EZ

ajh3, 2018-09-24:
Using this keymap, on Ubuntu: https://configure.ergodox-ez.com/layouts/XEL4/latest/0

Mouse movement has several issues. 

-Pressing down+right at the same time causes the cursor to lock in place. 
-Pressing down+left at the same time causes the cursor to move left only.
-Pressing up+right at the same time causes the cursor to move up only.
-Pressing up+left at the same time is the only diagonal combination that works reliably as expected.

The above keymap places the mouse on a momentary-accessed layer, but I've tested this on the main layer and the behavior is exactly the same. 

Side note, I recently tried an Ultimate Hacking Keyboard and the code they are using for mouse movement is far, far smoother than what QMK is using, and doesn't have any of these bugs, so that may be good to research.

I am not very experienced in C programming, although if someone can provide some suggestions on where to start debugging this, I am open to putting in some work to improve it.

noroadsleft, 2018-09-24:
#144 looks like it could be related, but I can't actually worked out if this was for sure solved/fixed.

Tagging @ezuk in hopes he can be of more help.

drashna, 2018-09-24:
There was actually a fix for this about 3 months ago, actually.  I don't think the Ergodox EZ Configurator has that commit. 

If you grab the C file from there and compile from source, it may fix the issue that you're seeing.

noroadsleft, 2018-09-25:
Also

https://config.qmk.fm/#/ergodox_ez/LAYOUT_ergodox

fdidron, 2018-09-25:
@ajh3 The Online configurator is indeed lagging behind, we are going to update the QMK dependency in the coming weeks. In the meantime, I compiled your layout against QMK's master branch and as @noroadsleft  and @drashna mentioned, it fixed the issue. You can download the hex file and modified source file from here: [aaron_qmk_master.zip](https://github.com/qmk/qmk_firmware/files/2413709/aaron_qmk_master.zip)

ajh3, 2018-09-25:
Thanks everybody for the help! I got the build environment set up and was able to build the firmware too based on your changes @fdidron. Would you mind explaining why it is required to change KEYMAP to LAYOUT_ergodox in the newer version? 

Is there a difference feature-wise in that QMK.fm configuration tool vs. the Ergodox EZ configurator (aside from more up-to-date dependencies), or is the latter just a prettier UI?

drashna, 2018-10-05:
> Is there a difference feature-wise in that QMK.fm configuration tool vs. the Ergodox EZ configurator (aside from more up-to-date dependencies), or is the latter just a prettier UI?

Basically that.  There are some other differences, like being able to set the RGB colors, which the QMK Configurator doesn't support, currently.

drashna, 2018-10-05:
@ajh3 would you mind if I closed this?

ajh3, 2018-10-05:
@drashna Go right ahead, and thanks for the info. For future reference, is this the right place to open issues about the Ergodox EZ specifically?

ezuk, 2018-10-08:
@ajh3 can always just ping us over email, contact@ergodox-ez.com -- we're pretty quick over there :)

drashna, 2018-10-08:
@ajh3 for general QMK stuff, yes.  And if you're using the QMK Configurator. And general issues.

If you're using the Ergodox EZ Configurator, then ... well, both here and contacting ezuk or whomever with the email listed above is a good idea.

closed 2018-10-26 by ajh3