qmk/qmk_firmware #3981: [ErgoDone] error reading intel hex file

homocomputeris, 2018-09-26:
I cannot flash ErgoDone in Arch:
```sh
./hid_bootloader_cli -mmcu=atmega32u4 ergodone_default.hex 
error reading intel hex file "ergodone_default.hex"
```

The build seems to be fine:
```shell
make clean && make ergodone:default
QMK Firmware 0.6.123
Deleting .build ... done
QMK Firmware 0.6.123
WARNING:
 Some git sub-modules are out of date or modified, please consider running:
 make git-submodule
 You can ignore this warning if you are not compiling any ChibiOS keyboards,
 or if you have modified the ChibiOS libraries yourself. 

Making ergodone with keymap default

avr-gcc (GCC) 8.2.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/ergodone/twimaster.c                                                           [OK]
Compiling: keyboards/ergodone/matrix.c                                                              [OK]
Compiling: keyboards/ergodone/expander.c                                                            [OK]
Compiling: keyboards/ergodone/ergodone.c                                                            [OK]
Compiling: keyboards/ergodone/keymaps/default/keymap.c                                              [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/keymap_common.c                                                                  [OK]
Compiling: quantum/keycode_config.c                                                                 [OK]
Compiling: quantum/process_keycode/process_unicode.c                                                [OK]
Compiling: quantum/process_keycode/process_unicode_common.c                                         [OK]
Compiling: tmk_core/common/host.c                                                                   [OK]
Compiling: tmk_core/common/keyboard.c                                                               [OK]
Compiling: tmk_core/common/action.c                                                                 [OK]
Compiling: tmk_core/common/action_tapping.c                                                         [OK]
Compiling: tmk_core/common/action_macro.c                                                           [OK]
Compiling: tmk_core/common/action_layer.c                                                           [OK]
Compiling: tmk_core/common/action_util.c                                                            [OK]
Compiling: tmk_core/common/print.c                                                                  [OK]
Compiling: tmk_core/common/debug.c                                                                  [OK]
Compiling: tmk_core/common/util.c                                                                   [OK]
Compiling: tmk_core/common/eeconfig.c                                                               [OK]
Compiling: tmk_core/common/report.c                                                                 [OK]
Compiling: tmk_core/common/avr/suspend.c                                                            [OK]
Compiling: tmk_core/common/avr/timer.c                                                              [OK]
Compiling: tmk_core/common/avr/bootloader.c                                                         [OK]
Assembling: tmk_core/common/avr/xprintf.S                                                           [OK]
Compiling: tmk_core/common/bootmagic.c                                                              [OK]
Compiling: tmk_core/common/mousekey.c                                                               [OK]
Compiling: tmk_core/protocol/lufa/lufa.c                                                            [OK]
Compiling: tmk_core/protocol/usb_descriptor.c                                                       [OK]
Compiling: tmk_core/protocol/lufa/outputselect.c                                                    [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Class/Common/HIDParser.c                                       [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Device_AVR8.c                                        [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/EndpointStream_AVR8.c                                [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Endpoint_AVR8.c                                      [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Host_AVR8.c                                          [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/PipeStream_AVR8.c                                    [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Pipe_AVR8.c                                          [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBController_AVR8.c                                 [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBInterrupt_AVR8.c                                  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/ConfigDescriptors.c                                       [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/DeviceStandardReq.c                                       [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/Events.c                                                  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/HostStandardReq.c                                         [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/USBTask.c                                                 [OK]
Linking: .build/ergodone_default.elf                                                                [OK]
Creating load file for flashing: .build/ergodone_default.hex                                        [OK]
Copying ergodone_default.hex to qmk_firmware folder                                                 [OK]
Checking file size of ergodone_default.hex                                                          [OK]
 * File size is fine - 18762/28672 (9910 free)

```
Tools
```
local/avr-binutils 2.31.1-1
local/avr-gcc 8.2.0-1
local/avr-libc 2.0.0-1
local/arm-none-eabi-binutils 2.31.1-1
local/arm-none-eabi-gcc 8.2.0-1
local/arm-none-eabi-newlib 3.0.0.20180831-1
local/dfu-programmer 0.7.2-1
local/dfu-util 0.9-1
```
Summon @yuhe00

MxBlu, 2018-09-26:
IIRC, this is a bug to do with avr-gcc 8.2.0: https://github.com/qmk/qmk_firmware/issues/3657

drashna, 2018-09-26:
Yeah, it's an avr-gcc 8.2 specific issue with the Pro Micro or caterina bootloader.

closed 2018-10-22 by drashna