qmk/qmk_firmware #3993: Planck Rev. 6 Make Error

Clevi-Villa, 2018-09-28:
Newbie here.
Here is the code.

$ make planck/rev6:default
Making planck/rev6 with keymap default

/bin/sh: arm-none-eabi-gcc: command not found
tmk_core/chibios.mk:89: lib/chibios/os/hal/lib/streams/streams.mk: No such file or directory
make[1]: *** No rule to make target 'lib/chibios/os/hal/lib/streams/streams.mk'.  Stop.
Make finished with errors
make: *** [Makefile:544: planck/rev6:default] Error 1

What could be causing the issue?
How do I fix it.

drashna, 2018-09-28:
re-run the installing script.   It's missing the arm gcc compiler.

Eg, run `util/qmk_install.sh`, and agree to everything.

Clevi-Villa, 2018-09-28:
No dice.

Mm... I believe I should be concerned about this...
' //
'qmk_driver_installer.exe' is not recognized as an internal or external command,
operable program or batch file.
~/qmk_utils /c/Users/Villa/Desktop/qmk_firmware

The AVR toolchain is not installed.
This is needed for building AVR based keboards.
Do you want to install it? (Y/N) y
/c/Users/Villa/Desktop/qmk_firmware/util/msys2_install.sh: line 16: wget: command not found

7-Zip [64] 16.02 : Copyright (c) 1999-2016 Igor Pavlov : 2016-05-21
p7zip Version 16.02 (locale=en_US.UTF-8,Utf16=on,HugeFiles=on,64 bits,4 CPUs x64)

Scanning the drive for archives:

ERROR: No more files
avr8-gnu-toolchain-installer-3.5.4.91-win32.any.x86.exe

System ERROR:
Unknown error -2147024872
rm: cannot remove 'avr8-gnu-toolchain-installer-3.5.4.91-win32.any.x86.exe': No such file or directory
warning: mingw-w64-x86_64-avrdude-6.3-2 is up to date -- skipping
 there is nothing to do

The ARM toolchain is not installed.
This is needed for building ARM based keyboards.
Do you want to install it? (Y/N) y
/c/Users/Villa/Desktop/qmk_firmware/util/msys2_install.sh: line 23: wget: command not found
unzip:  cannot find or open gcc-arm-none-eabi.zip, gcc-arm-none-eabi.zip.zip or gcc-arm-none-eabi.zip.ZIP.
rm: cannot remove 'gcc-arm-none-eabi.zip': No such file or directory
/c/Users/Villa/Desktop/qmk_firmware
source ~/qmk_utils/activate_msys2.sh

The line source ~/qmk_utils/activate_msys2.sh is already added to your /.bashrc
Not adding it twice!

******************************************************************************
Installation completed!
Please close this Window and restart MSYS2 MinGW'
//

_Unknown Error  -2147024872..._

drashna, 2018-09-28:
Sounds like you may have a bad install of MSYS. 

Though, try running `pacman --needed -S wget` and then rerun the setup script.

drashna, 2018-10-05:
Did you ever get this resolved?

Clevi-Villa, 2018-10-05:
Got occupied w/ personal affairs.

I am attempting to reinstall my OS, and will report back after a fresh install of everything, including MSYS2.

A corrupted install may have been the problem.

martin-damien, 2018-10-23:
Did you tryed `make git-submodule`?

I had such an error and it was because my submodules were out of date.

closed 2019-11-21 by drashna

ejahglass, 2020-06-17:
I'm having the exact same issue, but it doesn't make sense because it was working fine for me and then my most recent compile attempt i got

 "/bin/sh: arm-none-eabi-gcc: command not found
sh: arm-none-eabi-gcc: command not found
sh: arm-none-eabi-gcc: command not found
make[1]: *** [tmk_core/rules.mk:263: gccversion] Error 127
Make finished with errors
"