qmk/qmk_firmware #4070: Auto shift does not work for non-us keys

ant-i-c-s, 2018-10-04:
**Keyboard:** ErgoDox EZ

Auto shift works fine for most keys except `KC_NONUS_BSLASH`, which repeats instead of sending `|`.

drashna, 2018-10-04:
If you could, and you're compiling from source, could you apply this? 
```diff
diff --git a/quantum/process_keycode/process_auto_shift.c b/quantum/process_keycode/process_auto_shift.c
index 01d99445b..0d0930ee6 100644
--- a/quantum/process_keycode/process_auto_shift.c
+++ b/quantum/process_keycode/process_auto_shift.c
@@ -173,6 +173,8 @@ bool process_auto_shift(uint16_t keycode, keyrecord_t *record) {
       case KC_DOT:
       case KC_SLSH:
       case KC_GRAVE:
+      case KC_NONUS_BSLASH:
+      case KC_NONUS_HASH:
 #endif

         autoshift_flush();
```

Save this content to a file, and then use `git apply (filename)` to apply this, and see if it works. 

This may not work, though.

ant-i-c-s, 2018-10-04:
That worked! Thank you.

drashna, 2018-10-04:
Awesome!  I'll make a PR to merge this then.  Unless you want to :)

PR #4074 (2018-10-07): Add non-US Hash and Backslash to AutoShift handling
Fixes #4070 by adding the non-US characters to the list in process_auto_shift.

Tested and confirmed that this fixes the issue.
closed 2018-10-07 by jackhumbert