qmk/qmk_firmware #4097: Abstract ARM board info

drashna, 2018-10-08:
Mainly prep for Proton C and the inevitable wave of conversions.

yiancar, 2018-10-12:
Are we talking about the mcuconf halconf etc?

drashna, 2018-10-12:
Yes. 

Specifically, the `bootloader_defs.h`, `chconf.h`, `halconf.h`, `mcuconf.h` files, and the `boards` folder which needs to be added to the keyboard's folder, on top of the normal stuff.

I think @jackhumbert mentioned handling it like the drivers, so that if the specific controller is used, that it includes the correct folder (with all of the matching files) into the "include" path. 

This way, when we have people adding ARM versions of their boards to the repo, they don't have to worry about all of the extra files, where to find them, and that they're correct.  All they have to do is set the MCU info in the rules.mk, and that's it.

yiancar, 2018-10-12:
Well `halconf.h` is application specific, it tells the hal drivers which parts to enable. `chconf.h` is os application specific, selecting frequency theads etc. `mcuconfig.h` again are application specific for the peripherals. 

`bootloader_defs.h` is ok to move as those its actually only mcu specific.` Boards` can also be moved providing that all the pins are exposed.

yiancar, 2018-10-22:
SO any ideas on this or are people happy with the way it is now?

drashna, 2018-10-23:
well, "internal talk" is kind of leading towards an easier way to convert the pro micro boards over. 

The other option is maybe writing a guide on how to convert everything over.

yiancar, 2018-10-23:
I wouldnt mind writing a guide on how the files work. However there is a huge list of options. I can cover most of them. Some of the options go very deep regarding priorities and IRQs etc.

In general I think a guide can explain how to port over a keyboard to ARM with the basic options. SPI/I2C are pretty  straight forward.

drashna, 2018-10-23:
That would be nice! 

Though a guide on converting Pro Micro based boards to the Proton C would be best, right now.   Since I foresee that happening a lot.

yiancar, 2018-10-23:
Well I dont have a proton C but i imagine if someone setups a script to copy all the mcu/os files to a new directory, that would be it! matrix.c now works for arm as well so just redefining the pins should do

drashna, 2018-10-23:
Yeah, I think that was talked about as well. 

The real question is how to handle Pro Micro vs Proton C for boards.

yiancar, 2018-10-23:
well if you select an avr target it will build for pro micro if you select stm32 it will build for proton c

you just need the correct pins

drashna, 2019-02-25:
Jack added this.

closed 2019-02-25 by drashna