qmk/qmk_firmware #4120: [gccversion] Error 127

shinyamizuno1008, 2018-10-11:
1, I run the script to use my fourier keybord.
```
$ ./util/qmk_install.sh
$ make fourier:valgrahf
```
2, However, I got the err.
```
$ make fourier:valgrahf
QMK Firmware 0.6.137
Making fourier/rev1 with keymap valgrahf

/bin/sh: avr-gcc: command not found
sh: avr-gcc: command not found
sh: avr-gcc: command not found
make[1]: *** [gccversion] Error 127
make: *** [fourier:valgrahf] Error 1
```
3. I refer to [this reddit article](https://www.reddit.com/r/olkb/comments/5i4983/planckosxsh_avrgcc_command_not_found_gccversion/) which is same error as me.
I run the script which is written in the article.
```
$ brew reinstall avr-gcc@7
$ brew install crosspack-avr
$ brew update
$ brew upgrade
$ brew link avr-gcc
$ ./util/qmk_install.sh
```
```
the script above's outputs

$ ./util/qmk_install.sh

Updating Homebrew...
==> Auto-updated Homebrew!
Updated 2 taps (homebrew/core and homebrew/cask).
==> Updated Formulae
crosstool-ng            freetds                 hugo                    node-build              php                     sslmate                 twine-pypi
firebase-cli            geos                    jenkins                 openvdb                 php@7.1                 swiftformat             wakatime-cli

Already up-to-date.
Warning: px4/px4/gcc-arm-none-eabi 20171218 is already installed and up-to-date
To reinstall 20171218, run `brew reinstall gcc-arm-none-eabi`
Warning: dfu-programmer 0.7.2 is already installed and up-to-date
To reinstall 0.7.2, run `brew reinstall dfu-programmer`
Warning: avrdude 6.3_1 is already installed and up-to-date
To reinstall 6.3_1, run `brew reinstall avrdude`
==> Installing avr-gcc@7 from osx-cross/avr
==> Downloading https://gcc.gnu.org/pub/gcc/releases/gcc-7.3.0/gcc-7.3.0.tar.xz
Already downloaded: /Users/shinyamizuno/Library/Caches/Homebrew/downloads/b33c749ed9ba192d91270f50daf9ffbfc1359a6cdbd232b0c35b3d7f92f5b52e--gcc-7.3.0.tar.xz
==> Patching
patching file gcc/graphite.h
==> ../configure --target=avr --prefix=/usr/local/Cellar/avr-gcc@7/7.3.0 --libdir=/usr/local/Cellar/avr-gcc@7/7.3.0/lib/avr-gcc/7 --enable-languages=c,c++ --with-ld=/us
==> make
==> make install
==> Downloading https://download.savannah.gnu.org/releases/avr-libc/avr-libc-2.0.0.tar.bz2
Already downloaded: /Users/shinyamizuno/Library/Caches/Homebrew/downloads/02c166efff9cac7ac9108da0a404d5ab73732ab0d785909ed7bc76e2e6b953da--avr-libc-2.0.0.tar.bz2
==> ./configure --build=i686-apple-darwin17.7.0 --prefix=/usr/local/Cellar/avr-gcc@7/7.3.0 --host=avr
==> make install
==> Caveats
avr-gcc@7 is keg-only, which means it was not symlinked into /usr/local,
because it might interfere with other version of avr-gcc. This is useful if you want to have multiple version of avr-gcc installed on the same machine.

If you need to have avr-gcc@7 first in your PATH run:
  echo 'export PATH="/usr/local/opt/avr-gcc@7/bin:$PATH"' >> ~/.bash_profile

For compilers to find avr-gcc@7 you may need to set:
  export LDFLAGS="-L/usr/local/opt/avr-gcc@7/lib"

==> Summary
🍺  /usr/local/Cellar/avr-gcc@7/7.3.0: 1,682 files, 175.3MB, built in 31 minutes

------------------------------------------------------------------------------------
$ brew install crosspack-avr

Updating Homebrew...
==> Auto-updated Homebrew!
Updated 1 tap (homebrew/cask).
No changes to formulae.

Error: No available formula with the name "crosspack-avr"
==> Searching for a previously deleted formula (in the last month)...
Warning: homebrew/core is shallow clone. To get complete history run:
  git -C "$(brew --repo homebrew/core)" fetch --unshallow

Error: No previously deleted formula found.
==> Searching for similarly named formulae...
Error: No similarly named formulae found.
==> Searching taps...
==> Searching taps on GitHub...
Error: No formulae found in taps.

------------------------------------------------------------------------------------
$ brew update

Already up-to-date.

------------------------------------------------------------------------------------
$ brew upgrade
```
**What should I do? Please help.**

```
my info 
OS:            Mac OSX ver 10.13.6
keyboard: fourier  (keebio)

```

drashna, 2018-10-11:
Try running `  echo 'export PATH="/usr/local/opt/avr-gcc@7/bin:$PATH"' >> ~/.bash_profile`, and reopen the terminal.

shinyamizuno1008, 2018-10-11:
@drashna 
I'm so stupid. done. Thank you !
```
$ make fourier:valgrahf:avrdude
QMK Firmware 0.6.137
Making fourier/rev1 with keymap valgrahf and target avrdude

avr-gcc (GCC) 7.3.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text    data     bss     dec     hex filename
      0   21224       0   21224    52e8 .build/fourier_rev1_valgrahf.hex

Compiling: tmk_core/common/command.c                                                                [OK]
Linking: .build/fourier_rev1_valgrahf.elf                                                           [OK]
Creating load file for flashing: .build/fourier_rev1_valgrahf.hex                                   [OK]
Checking file size of fourier_rev1_valgrahf.hex                                                     [OK]
 * The firmware size is fine - 21224/28672 (7448 bytes free)
Copying fourier_rev1_valgrahf.hex to qmk_firmware folder                                            [OK]
Detecting USB port, reset your controller now.........
Detected controller on USB port at /dev/tty.usbmodem1421

Connecting to programmer: .
Found programmer: Id = "CATERIN"; type = S
    Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:
    Device code: 0x44

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9587 (probably m32u4)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file ".build/fourier_rev1_valgrahf.hex"
avrdude: input file .build/fourier_rev1_valgrahf.hex auto detected as Intel Hex
avrdude: writing flash (21224 bytes):

Writing | ################################################## | 100% 1.63s

avrdude: 21224 bytes of flash written
avrdude: verifying flash memory against .build/fourier_rev1_valgrahf.hex:
avrdude: load data flash data from input file .build/fourier_rev1_valgrahf.hex:
avrdude: input file .build/fourier_rev1_valgrahf.hex auto detected as Intel Hex
avrdude: input file .build/fourier_rev1_valgrahf.hex contains 21224 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.19s

avrdude: verifying ...
avrdude: 21224 bytes of flash verified
```

closed 2018-10-11 by shinyamizuno1008

drashna, 2018-10-11:
Awesome, glad to heat that this fixed the issue!  

And yeah, the AVR GCC thing has been an issue for a bit now.