qmk/qmk_firmware #4152: error: stray '/342" in program AND error: expected....

lilacmoonflower, 2018-10-17:
Total noob here. I have no experience coding at all but managed to do some troubleshooting to work on my keymap.c files. Keymapping is pretty simple but I am having some issues with creating macros. Here is a pastebin showing the errors I am getting when compiling the firmware:

https://pastebin.com/Uu1XKD5c

Looks like there is an error for every macro I wrote. Empirical evidence that I suck at this.. Here are the errors I am getting:
error: stray '/342' in program 
(same errors for '/200' and '/213')
error: expected ':' or "..." before 'send_string_P'

Here is a pastebin of the new code I added to keymap.c: https://pastebin.com/JkuGWaA5 #

drashna, 2018-10-17:
Well, it compiles fine on my end. :D

But could you post the entire keymap.c file? 

Chances are that you edited out something that you shouldn't have.

lilacmoonflower, 2018-10-17:
Hi Drashna! Thank you for the fast response!

Here is a pastebin of my full keymap.c: https://pastebin.com/Z1qjNBHM

Basically, everything is the same with the exception of the new keymap I modified (NUMLOCK, GAMEPAD, and MACROS) and I added the code above everything. 

I also noticed that when I use either MO or TT to toggle the MACROS layer, it acts as a normal toggle and doesn't go back to the GAMEPAD layer. Same issue with TG even if I map a toggle back to the GAMEPAD.

I appreciate the support!

lilacmoonflower, 2018-10-17:
Thanks! I did another test using your cleaner code and added the defines after enum. I am still getting the "error: expected ..." for each macro and process_record_keymap. Could you please help me take another look?

error: https://pastebin.com/uWUsnjGs
keymap: https://pastebin.com/FCLvbCCv

drashna, 2018-10-17:
Line 99 in the keymap.c file, you're missing a colon. 

You have: 
```c
case SMITE_TAKEIT
```
when you should have this: 
```c
case SMITE_TAKEIT:
```

drashna, 2018-10-17:
Well, for the MO (macro), you need transparent, not "NO" for it to trigger properly.  So on the macro layer, you need to change a couple of the spaces.

As for the TT, I don't see that being used, actually.  I know it's defined, but the TT doesn't appear to be called anywhere but the media layer.

lilacmoonflower, 2018-10-18:
Woot! Ok now it is 100% working. Thank you again for all the help!

closed 2018-10-18 by lilacmoonflower

drashna, 2018-10-18:
Awesome!  Glad to hear it!