qmk/qmk_firmware #4153: Bin files created are extreamly large

shalzz, 2018-10-17:
Created a firmware image for any keymap of TADA68, which uses a bin format now suddenly results in unusually large firmware.

For eg: `make tada:unix:bin` or `make tada68:shalzz:bin` creates a bin file that is 8.1M in size.
Output:
```
 24K -rw-r--r-- 1 shalzz shalzz 8.1M Oct 17 13:05 tada68_shalzz.bin
300K -rwxr-xr-x 1 shalzz shalzz 298K Oct 17 13:05 tada68_shalzz.elf
 48K -rw-r--r-- 1 shalzz shalzz  47K Oct 17 13:05 tada68_shalzz.hex
224K -rw-r--r-- 1 shalzz shalzz 221K Oct 17 13:05 tada68_shalzz.map
 24K -rw-r--r-- 1 shalzz shalzz 8.1M Oct 17 13:17 tada68_unix.bin
300K -rwxr-xr-x 1 shalzz shalzz 298K Oct 17 13:12 tada68_unix.elf
 48K -rw-r--r-- 1 shalzz shalzz  47K Oct 17 13:12 tada68_unix.hex
220K -rw-r--r-- 1 shalzz shalzz 218K Oct 17 13:12 tada68_unix.map
```

drashna, 2018-10-17:
That's the issue.  There is a known bug with 8.2.  You'll need to roll back to 8.1 or 7.3.  Once you've done that, it should fix this issue.

shalzz, 2018-10-19:
That fixed it. Thanks

closed 2018-10-19 by shalzz

drashna, 2018-10-19:
Awesome, glad to hear it !