qmk/qmk_firmware #4175: Can't find a JSON default layout file to import for Massdrop ctrl keyboard

jamesalmeida, 2018-10-19:
I'm trying to use QMK Configurator to change my keyboard configuration but Im at a loss since it doesn't have a default layout to load and I can't find a JSON default layout file to import either. Anyone know how I can do this?

just-another-jxliu, 2018-10-19:
This one? https://github.com/qmk/qmk_firmware/blob/master/keyboards/massdrop/ctrl/info.json

noroadsleft, 2018-10-19:
@just-another-jxliu That file is for a different purpose; it's not for keymaps.

@jamesalmeida I can file a PR to the Configurator repo (it's not in qmk/qmk_firmware) that adds a default for the CTRL. Give me a bit.

drashna, 2018-10-19:
To clarify, the default keymap in the Configurator isn't handled by QMK Firmware. It's actually files added to the Configurator repo.   So there is a bit of a disconnect there.

jamesalmeida, 2018-10-19:
> This one? https://github.com/qmk/qmk_firmware/blob/master/keyboards/massdrop/ctrl/info.json

At first I thought this was it too but it's not the right file.

jamesalmeida, 2018-10-19:
> @just-another-jxliu That file is for a different purpose; it's not for keymaps.
> 
> @jamesalmeida I can file a PR to the Configurator repo (it's not in qmk/qmk_firmware) that adds a default for the CTRL. Give me a bit.

Thank you!

drashna, 2018-10-19:
@jamesalmeida Specifically, you'd need to add a layout here: https://github.com/qmk/qmk_configurator/tree/master/keymaps

Once that's done, then the "load default" button will work.

noroadsleft, 2018-10-19:
PR is up. :arrow_up:

noroadsleft, 2018-10-19:
PR is merged.

https://config.qmk.fm/#/massdrop/ctrl/LAYOUT

drashna, 2018-10-20:
Awesome.

Closing this issue since it should be resolved.

closed 2018-10-20 by drashna