qmk/qmk_firmware #4185: "Soft meta" feature

eraserhd, 2018-10-21:
I've implemented a kind of software Meta key for my ErgoDox mapping, where holding down `X` or `,` acts as though any keys pressed while they are held are preceded by tapping Escape.  This has the advantage that it works regardless of how Option keys are configured on Mac OS, meaning I can carry around my keyboard to other people's computers, and also that I can work around mapping bugs in Alacritty.

What I have is kind of hacked-together, but if there is enough interest, I'll make it a proper feature and contribute it back.

(What I have is [here](https://github.com/eraserhd/qmk_firmware/blob/master/keyboards/ergodox_ez/keymaps/eraserhd/soft_meta.h), by the way.)

drashna, 2019-11-21:
Well, Ctrl-GUI swap was added recently, that may be a better option?

eraserhd, 2019-11-29:
I'm no longer using this, and the link is no longer valid, so closing.

closed 2019-11-29 by eraserhd