qmk/qmk_firmware #4196: layer buttons broken when in tty on linux

plattfot, 2018-10-21:
Hi,

I encountered a really strange issue when using the latest version of qmk_firmware on my ergodox infinity and ergotravel. The layer switching buttons doesn't work when I'm in a console (TTY) 

Works perfectly fine when in X11, but when switching to a TTY say ctrl+alt F2 it stops working so I cannot switch back as my F1-F12 are on another layer. Buttons that are using OSL() stops working where as buttons with MO() gets stuck on that layer. And I cannot exit that layer except either unplug it or switch to X11 using another keyboard (not running qmk). Which it then goes back to normal.

If I use a firmware based on qmk_firmware about 6 month ago (81ae0fb10) it works fine in both TTY and X11.

This seems really weird to me as from what I understand the layers should be internal to the keyboard. So I'm not sure how they get affected by the OS on my machine. Any ideas what could be wrong?

Using Arch linux with kernel: 4.18.14-arch1-1-ARCH
Firmware with broken TTY is based on ea17e363a

Thanks

vomindoraan, 2018-10-21:
What's your `keyboard:keymap`?

drashna, 2018-10-22:
Maybe try adding `#define STRICT_LAYER_RELEASE` to your config.h?

plattfot, 2018-10-22:
> What `keyboard:keymap` combo did you compile the firmware with? e.g. `ergodox_infinity:default`
Sorry forgot to mention that. Using my own keymap I have in my fork of this project: https://github.com/plattfot/qmk_firmware/tree/master/layouts/community/ergodox/plattfot
`ergodox_infinity:plattfot` or `ergotravel:plattfot` 

> Maybe try adding `#define STRICT_LAYER_RELEASE` to your config.h?
I'll try that when I come home tonight.

plattfot, 2018-10-23:
Tested `#define STRICT_LAYER_RELEASE` with both my ergodox infinity and my ergotravel. No difference. 

But my previous statement was wrong. There's nothing wrong with my ergotravel. Turns out that right alt doesn't work when I'm in the console. 

So this seems to only affect the ergodox infinity. Using MO() will work until I release it, then it locks my keyboard. OSL will lock my keyboard as soon as I press the key. 

Just jumping back to X11 will unfreeze it.

plattfot, 2018-10-23:
Ok,
I'll try disabling that and see if that helps.

Otherwise I'll see if I can do some binary searching to see if I can nail down the commit.

plattfot, 2018-10-24:
This is the commit I found when doing a git bisect: ade22f8e2c272044ea2f80ff6fe5ca9576858939
Which corresponds to tag 0.6.68, checking out 0.6.67 works.

Hope that helps.

drashna, 2018-10-24:
That's the commit that introduced the issue for you, correct? 

If so, ... yeah, that's .... ugh.  There were some changes to the ChibiOS stuff, which is what the ergodox infinity uses (as opposed to LUFA for AVR).  

Maybe, try removing this block, and see if that fixes the issue:
https://github.com/qmk/qmk_firmware/blob/b360a1c8d23671f17a985cda7a37caba83b6ca56/tmk_core/protocol/chibios/usb_main.c#L721-L727

plattfot, 2018-10-25:
> That's the commit that introduced the issue for you, correct?
Yes.

That block seems to be the cause of this issue.
Commented that out, recompiled and no more locking in the TTY.

So it has something to do with mouse emulation?

drashna, 2018-10-25:
Honestly, I'm not sure.  

Do you have MOUSEKEY_ENABLE set to yes in your keymap's rules.mk?

vomindoraan, 2018-10-25:
Good catch @drashna!

I'm not too familiar with ChibiOS, but it's possible that the suspended thread is never resumed due to a collision in the endpoint numbers (`*_EPNUM`). Actually, it's possible that [this line](https://github.com/qmk/qmk_firmware/blob/ee700b2e831067bdb7584425569b61bc6329247b/tmk_core/protocol/usb_descriptor.h#L61) might be the culprit. I don't see `MOUSE_ENABLE` defined anywhere.

drashna, 2018-10-25:
Hmm.....

If you replace the `MOUSE_ENABLE` with `MOUSEKEY_ENABLE` there, I wonder if that would fix the issue, as well. 

https://github.com/qmk/qmk_firmware/blob/ee700b2e831067bdb7584425569b61bc6329247b/tmk_core/protocol/usb_descriptor.h#L61-L66

drashna, 2018-10-25:
@jackhumbert is this section of code critical? 
https://github.com/qmk/qmk_firmware/blob/b360a1c8d23671f17a985cda7a37caba83b6ca56/tmk_core/protocol/chibios/usb_main.c#L721-L727

plattfot, 2018-10-25:
> Do you have MOUSEKEY_ENABLE set to yes in your keymap's rules.mk?

@drashna, I don't have any mouse related enabled in my rules.mk. I do have mouse keys mapped to one of my layers though.

drashna, 2018-10-25:
The Ergodox Infinity does have `MOUSEKEY_ENABLE = yes` by default.  So if you're not disabling it in your keymap explicitly, it's enabled.

If you're using the keys, that's ... problematic then.  If you're not, then disable the feature altogether, and that should also fix this issue.   At least until we can get an answer from Jack, or somebody to fix the relevant code.

drashna, 2018-10-25:
Alternatively, if you remove this chunk of code, do the mousekeys still work?

plattfot, 2018-10-25:
Hmm, good question. Didn't try my mousekeys after I removed that block.

I'll test that tonight and report back.

drashna, 2018-10-25:
Yes, please do! 

And test out as much of the mousekey functionality as you can.

If it works without that block, then it may be best to remove it from the master branch, so others don't run into this (or other issues).  But if it doesn't work, then .... a fix will be needed.

fredizzimo, 2018-10-25:
It looks like the OS might be blocking the mouse HID messages, and therefore locking the keyboard up. AVR has a timeout of 10ms, so the following code should do the same on ChibiOS

```C
void send_mouse(report_mouse_t *report) {
  osalSysLock();
  if(usbGetDriverStateI(&USB_DRIVER) != USB_ACTIVE) {
    osalSysUnlock();
    return;
  }

  if(usbGetTransmitStatusI(&USB_DRIVER, MOUSE_IN_EPNUM)) {
    /* Need to either suspend, or loop and call unlock/lock during
     * every iteration - otherwise the system will remain locked,
     * no interrupts served, so USB not going through as well.
     * Note: for suspend, need USB_USE_WAIT == TRUE in halconf.h */
    if (osalThreadSuspendTimeoutS(&(&USB_DRIVER)->epc[MOUSE_IN_EPNUM]->in_state->thread, MS2ST(10)==MSG_TIMEOUT){
      osalSysUnlock();
      return;
    }
  }
  usbStartTransmitI(&USB_DRIVER, MOUSE_IN_EPNUM, (uint8_t *)report, sizeof(report_mouse_t));
  osalSysUnlock();
}
```

Could you try replacing the function with that and see if it helps? 

There's also another more minor problem. We seem to send mouse reports even when you don't use the mouse keys at that moment, if that was fixed then it should lock up only when the mouse keys are used. It would also slightly improve the responsibility in the affected situation, even with the fixed code snipped above, as there's a delay of 10ms.

plattfot, 2018-10-25:
@fredizzimo, I'll add that to my to do list for tonight.

plattfot, 2018-10-26:
> Alternatively, if you remove this chunk of code, do the mousekeys still work?

@drashna, Mouse keys still works when the code block is disabled.

```c
void send_mouse(report_mouse_t *report) {
  osalSysLock();
  if(usbGetDriverStateI(&USB_DRIVER) != USB_ACTIVE) {
    osalSysUnlock();
    return;
  }

  if(usbGetTransmitStatusI(&USB_DRIVER, MOUSE_IN_EPNUM)) {
    /* Need to either suspend, or loop and call unlock/lock during
     * every iteration - otherwise the system will remain locked,
     * no interrupts served, so USB not going through as well.
     * Note: for suspend, need USB_USE_WAIT == TRUE in halconf.h */
    if (osalThreadSuspendTimeoutS(&(&USB_DRIVER)->epc[MOUSE_IN_EPNUM]->in_state->thread, MS2ST(10)==MSG_TIMEOUT)) {
      osalSysUnlock();
      return;
    }
  }
  usbStartTransmitI(&USB_DRIVER, MOUSE_IN_EPNUM, (uint8_t *)report, sizeof(report_mouse_t));
  osalSysUnlock();
}
```
This fixed the locking.

> There's also another more minor problem. We seem to send mouse reports even when you don't use the mouse keys at that moment, if that was fixed then it should lock up only when the mouse keys are [...]

@fredizzimo, How hard is it to fix so that this only applies to the mouse keys? 
Note that I don't feel the delay when I have drashna's patched applied.

drashna, 2018-10-26:
So the last code block that I posted worked fine, correct? 

If so, well that's already only enabled and called if Mousekeys is enable.   So if that last patch (the big one you've quoted) works, then it also only applies if mousekeys is eanbled, and we can then safely merge the code change (and probably help others that have been experiencing this).

plattfot, 2018-10-26:
Correct. I used the one in your branch `drashna:fix_mousekey_code_chibios`. 
Which as far as I can tell the same as you posted.

Switching layers works in the TTY, the mouse works in X11 and didn't notice any lag when in the TTY. So for me this patch fixed the issue I was having.

drashna, 2018-10-26:
Yup, that branch uses the exact change posted here.  

And I'm glad to hear that!   I'll see about getting it merged in.

plattfot, 2018-10-26:
Awesome!

Thanks for all the help!

PR #4237 (2018-10-27): Fix mouse_send() on chibiOS so it won't lock up the firmware
Fixes #4196
closed 2018-10-27 by jackhumbert