qmk/qmk_firmware #4200: Add support for Bootmagic Lite

drashna, 2018-10-22:


drashna, 2018-10-22:
Well, it's easy to "add" this, but the problem that I'm running into is that I have no idea how to specify which locations, other than using 4 values (space row and column, and then b row and column).

https://github.com/drashna/qmk_firmware/blob/bootmagic_lite/quantum/quantum.c#L947-L990

Wilba6582, 2018-10-23:
FWIW my `bootmagic_lite` is now 50% lighter, only needs top left (i.e. row 0 col 0) switch closed on power on. This makes it even easier to use in normal cases (fully built keyboards) and when PCB is unassembled (shorting just one switch's pads while connecting power). I know the original bootmagic needed two keys held down to make it less likely to happen accidentally, but for the "lite" use-case, the consequences of accidental triggering is minimal, in most compiled-keymap-QMK scenarios.

In my latest implementation, I break out the EEPROM reset (QMK core + keyboard specific) into a separate function, which can be called from `bootmagic_lite` or via a raw HID command (long term goal of having host trigger a new firmware upload process). So for me to use this common bootmagic_lite, I would need an `eeprom_reset` as weak linked function, so I could add my own EEPROM reset code. But then I'd probably still want to roll my own `bootmagic_lite` function. Just something to think about... maybe I'm not the only one that uses EEPROM outside the QMK core range.

Another observations: it should wait 2 x `DEBOUNCING_DELAY` (observed that 1x is not enough)

drashna, 2018-10-23:
Thanks for the feedback, and that makes things easier on me, actually.   Especially just the one key.  Having 2 defines is best, IMO, so thanks.

For the implementation, I defined `bootmagic_lite` weakly, so that could be replaced, but I take your point, and will add the eeprom_reset as well. 

Again, super helpful feedback, and thanks!

drashna, 2018-10-23:
Actually, I'm not sure that the eeprom reset is needed for most boards.

For the Zeal, I can totally understand, though.

I think I'll leave out the eeprom, and leave `bootmagic_lite` weak, so it can be replaced per board.