qmk/qmk_firmware #4209: Issues compiling ARM on Ubuntu 18.04

drashna, 2018-10-22:
THere are some issues with the ARM library on Ubuntu 18.04, that cause it to error out. 

Namely, this: 
https://github.com/bbcmicrobit/micropython/issues/514#issuecomment-404759614

Downloading the "deb" files from the next build fixes the issues. 

https://packages.ubuntu.com/cosmic/all/libnewlib-dev/download
https://packages.ubuntu.com/cosmic/all/libnewlib-arm-none-eabi/download

And running `sudo dpkg -i ....` fixes the issue though.

Do we want to and this to the build script for just 18.04? 

@jackhumbert @skullydazed @fredizzimo

MxBlu, 2019-02-21:
Would like to point out this is still valid a few months later 😆 WSL runs on 18.04 at the moment.

tprk77, 2019-05-22:
I ran into this yesterday. For others finding this issue, I got a bunch of errors like:

```
Linking: .build/massdrop_ctrl_default.elf                                                           [ERRORS]
 | 
 | /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/rdimon-crt0.o: Conflicting CPU architectures 13/1
 | /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/rdimon-crt0.o
 | /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: .build/massdrop_ctrl_default.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg.a(lib_a-atexit.o) does not
 | /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg.a(lib_a-atexit.o)
 | /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: .build/massdrop_ctrl_default.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg.a(lib_a-exit.o) does not
 | /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg.a(lib_a-exit.o)
 | /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: .build/massdrop_ctrl_default.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg.a(lib_a-fini.o) does not
 | /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg.a(lib_a-fini.o)
 | /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: .build/massdrop_ctrl_default.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg.a(lib_a-impure.o) does not
 | /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg.a(lib_a-impure.o)
 | /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: .build/massdrop_ctrl_default.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg.a(lib_a-init.o) does not
 | /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg.a(lib_a-init.o)
 | /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: .build/massdrop_ctrl_default.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg.a(lib_a-memcpy-stub.o) does not
 | /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg.a(lib_a-memcpy-stub.o)
 | /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: .build/massdrop_ctrl_default.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg.a(lib_a-memset.o) does not
 | /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg.a(lib_a-memset.o)
 | /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: .build/massdrop_ctrl_default.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg.a(lib_a-rand.o) does not
 | /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg.a(lib_a-rand.o)
```

The [upstream Ubuntu bug](https://bugs.launchpad.net/ubuntu/+source/newlib/+bug/1767223) has not been fixed.

Installing the new 18.10 packages resolved the issue.

**Before:**

```
$ dpkg -l "*newlib*" | grep ^ii | head -2
ii  libnewlib-arm-none-e 2.4.0.20160527- all             ...
ii  libnewlib-dev        2.4.0.20160527- all             ...
```

**After:**

```
$ dpkg -l "*newlib*" | grep ^ii | head -2
ii  libnewlib-arm-none-e 3.0.0.20180802- all             ...
ii  libnewlib-dev        3.0.0.20180802- all             ...
```

And now everything seems to be compiling fine.

It would have been nice if the setup script somehow took care of this. But I can't really think of a nice way to do that, since it involves downloading some "random" packages without signatures, etc.

Maybe the setup script could print a warning message?

slashfoo, 2020-07-14:
I'm facing this issue while using debian 10, it uses newlib 3.3.0.

Would there be a way to work around this (easily) using LIBRARY_PATH or other environment variables?

oars, 2021-04-20:
Same issue as @slashfoo when compiling for the moonlander

```
$ ~/qmk_firmware ±firmware20 » dpkg -l "*newlib*" | grep ^ii | head -2
ii  libnewlib-arm-none-eabi        3.3.0-1           all          C library and math library compiled for bare metal using Cortex A/R/M
ii  libnewlib-dev                  3.3.0-1           all          C library and math library intended for use on embedded systems
```

uqs, 2021-08-07:
Sorry for not helping much, but I too have build problems on Debian with 3.3.0-1 installed. I'm tying to build the default preonic/rev3 keymap.

bbfsdev, 2021-11-28:
Any updates on this issue? Workarounds?

MxBlu, 2021-11-28:
> Any updates on this issue? Workarounds?

The workaround is in the opening post - download the .deb and install it directly (at your own risk, idk how much things have diverged since then).

bbfsdev, 2021-11-30:
Hello all, I used the following instructions to install the package:
1. https://askubuntu.com/questions/1243252/how-to-install-arm-none-eabi-gdb-on-ubuntu-20-04-lts-focal-fossa
2. Used this version: https://developer.arm.com/-/media/Files/downloads/gnu-rm/10.3-2021.10/gcc-arm-none-eabi-10.3-2021.10-x86_64-linux.tar.bz2

And could successfully build on debian.

tzarc, 2022-06-18:
Workaround already known in the opening post, closing.

closed 2022-06-18 by tzarc