qmk/qmk_firmware #4213: [help] Teensy 3.2 and I2C

FromtonRouge, 2018-10-22:
Hello, it's not an issue but I need some help with Teensy 3.2 and I2C.

I'm working on a split orthographic steno keyboard with a Teensy 3.2 on the left side and a mcp23018 on the right side. The left side is ok but I can't find a way to communicate with the mcp23018 when using the ChibiOS I2C library.

When I do the following in my matrix_init():

`palSetPadMode(TEENSY_PIN18_IOPORT, TEENSY_PIN18, PAL_MODE_OUTPUT_OPENDRAIN); // SDA
palSetPadMode(TEENSY_PIN19_IOPORT, TEENSY_PIN19, PAL_MODE_OUTPUT_OPENDRAIN); // SCL
i2cStart(&I2CD1, &i2c_config);
tx[0] = MCP_IODIRA;
tx[1] = 0b01111111;
i2cMasterTransmitTimeout(&I2CD1, MCP_I2C_ADDR_WRITE, tx, 2, NULL, 0, MS2ST(500));
`
(Clock in i2c_config is set to 400000 Hz)

The timeout is triggered and the I2C driver is in I2C_LOCKED state.
I have external pullups on SDA and SCL (2.2 kOhms). The circuitry seems to work because I did a simple test in Arduino and the Wire library and it worked well.

I can't find an example of I2C with a Teensy3.1/3.2 in the qmk repository or in ChibiOS-Contrib. Any help is appreciated, thx.

drashna, 2018-10-22:
Try setting the frequency to 100000 HZ.  I know the "split common" code needs that on some boards.

FromtonRouge, 2018-10-23:
I got the same behavior with 100000 Hz :(

nicolai86, 2019-03-10:
@FromtonRouge did you figure this out?

FromtonRouge, 2019-03-10:
Nope 😕 but I changed my keyboard design, it's not splitted anymore

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc