qmk/qmk_firmware #4229: using media keys

Jungwoo0213, 2018-10-24:
Hello! I am currently using Satan gh60 and I am using Windows 10.

KC_MEDIA_NEXT_TRACK | KC_MNXT | Next Track (Windows)
KC_MEDIA_PREV_TRACK | KC_MPRV | Previous Track (Windows)
KC_MEDIA_STOP | KC_MSTP | Stop Track (Windows)

I used these media keys using the aliases like KC_MSTP on my code, but it does not interact with any of the media programs, like foobar2000, Groove Music or youtube.

So I looked up the Autohotkey history and it shows that the registered media keys are "artificial" which I have no idea what it means. I think this has to do with this because if I register media keys with keys that were redirected by Autohotkeys it does not say artificial and it works fine.

drashna, 2018-10-24:
Odd.   If you use Switch Hitter, I get this for the "play" button:
```
7:41.0108  (0xB3, BIOS 0xE000) DOWN
17:41.0229  (0xB3, BIOS 0xE000) UP -> 121ms
```

And it works properly.  It may be worth testing this out.

drashna, 2018-10-24:
It sounds like AHK is intercepting it, and messing with it somehow.  :(

Jungwoo0213, 2018-10-24:
As I have tried to open and close various programs, I turned out that I was not AHK.
It was actually the Google Music extension in Chrome browser.
I removed it and it works fine now!

max-su, 2019-02-11:
Thanks so much! This helped me out a lot I had the same issue on Windows but not on Mac. 👍 also worked for me.

closed 2019-11-21 by drashna

Xiseous, 2024-08-18:
Reopening, W11,

KC_MEDIA_PLAY_PAUSE	KC_MPLY	Play/Pause Track

Not working using full name or alias,

Previous, Next, Volume up and down interact without issues.