qmk/qmk_firmware #4254: Re-Unify Split Keyboard Code

drashna, 2018-10-27:
Specifically, @mtei and @MakotoKurauchi has done some awesome work to expand the functionality of the serial code for some of the split keyboards.

However, this has happened independently and outside of the Split Common code.  Meaning that the improvements made have not improved the split code, and have cause it to diverge further, which was the entire point of the Split Common code: to remove the heavily fragmentation that was happening, and make the code easier to maintain for all (most) split keyboards. 

Additionally, the Orthodox and BFO9000 are using customized matrix code to allo for more than 8 columns.  This should be ported over to the Split Common code, as well. 

This isn't to attack or berate them for their work.  It's pointing out a big issue that I see.  

Ideally, these two sets of files need to be merged together, ind a way that works for everyone.

[ ] Configurable Serial Pin
[ ] Support both I2C and Serial at the same time that works with the configurator
[ ] Soft Serial? 
[ ] Support more than 8 columns/rows

Also tagging @That-Canadian and @nooges about this, in the hopes that they have some input/can help hout here.

Lenbok, 2018-10-28:
In most cases converting keyboards over to use the split common code is fairly straight forward, but just needs someone that actually has one of the boards to do a final test.

Perhaps when someone submits any PR targetting one of these boards, ask them to first convert the board to split common (since presumably they have one of the boards to test)?

drashna, 2018-10-28:
Testing is definitely ideal.  However, some of the code for the helix code has diverged greatly from the split code, in general (eg, that of the lets Split). So this isn't just a small project.  And getting it to work with bot sets (basically) will absolutely need testing. 

And the other factor is how to handle the configurator.

mtei, 2018-10-30:
helix/serial.c has enough backward compatibility. Please look at the sample. #4293

The problem is that split_common/serial.c is not configurable. I think that serial.c should have been made configurable when quantum/split_common was created.

drashna, 2018-10-30:
@mtei okay, good to know.

And absolutely agree with you.  

Unfortunately, I'm definitely still learning (though, that's a good thing, too, as there is always more to learn!), and didn't catch that, at the time. 

To update the code would be more of a "merge" than a copy job.

Additionally, one of the issues that I foresee here is the "use serial", "use i2c" paradigm.  Right now, the split common code is configured in such a way that it's really one or the other, but I know that won't work well for how the Helix (and similar boards) work.

The other issue with this paradigm is that it has issues on the Configurator.  Basically, anything that relies on the `config.h` or `rules.mk` of the keymap folder is not usable on the Configurator, and may cause the build process to fail.

mtei, 2018-11-01:
helix-serial.c supports the following APIs compatible with let's split.
https://github.com/qmk/qmk_firmware/blob/e18eab7a81c51015c8bd81931f90cf1dc808ac78/keyboards/lets_split/serial.h-old#L15-L25

helix-serial.c also supports a new API.
https://github.com/qmk/qmk_firmware/blob/e18eab7a81c51015c8bd81931f90cf1dc808ac78/keyboards/lets_split/serial.h#L44-L69

The let's split compatible API is implemented as a wrapper function in the following part on helix-serial.c.
https://github.com/qmk/qmk_firmware/blob/e18eab7a81c51015c8bd81931f90cf1dc808ac78/keyboards/lets_split/serial.c#L73-L116

I think the wrapper function on helix-serial.c should be moved to matrix.c (or a new file).  By so doing, serial.c can be immutable without being affected by changes in the caller's source (matrix.c or other source files).  Also, it is possible to exclude `SERIAL_SLAVE_BUFFER_LENGTH` and `SERIAL_ MASTER_BUFFER_LENGTH` from config.h, and the setting is simplified.

mtei, 2018-11-06:
In #4379, I tried solving the following problem.

* Configurable Serial Pin
* The data structure of the master-slave communication was made more freely.
* Support both I2C and Serial at the same time.

mtei, 2018-11-16:
## draft: new configuration  about use i2c, serial

### master-slave communication use serial case
In keyboards/KEYBOARD/rules.mk 
```makefile
SPLIT_KEYBOARD = yes
SPLIT_COMMUNICATION = serial
OLED_ENABLE = yes # or no
# if need   SOME_DEVICE_ENABLE = yes or no it's use I2C
```
In keyboards/KEYBOARD/config.h
```c
#define SOFT_SERIAL_PIN D0 // or D1, D2, D3, E6
```
### master-slave communication use I2C case
In keyboards/KEYBOARD/rules.mk 
```makefile
SPLIT_KEYBOARD = yes
SPLIT_COMMUNICATION = i2c
OLED_ENABLE = yes # or no
# if need   SOME_DEVICE_ENABLE = yes or no it's use I2C
```
In keyboards/KEYBOARD/config.h
```c
/* nothing */
```

### qmk_firmware/common_features.mk implementation
```makefile
ifeq ($(strip $(SPLIT_KEYBOARD)), yes)
  #
  # SPLIT_COMMUNICATION and OLED_ENABLE settings convert to C lang Macros
  #
  USE_I2C := no
  SPLIT_SRC := $(QUANTUM_DIR)/split_common/split_flags.c
  SPLIT_SRC +=  $(QUANTUM_DIR)/split_common/split_util.c 

  ifeq ($(strip $(SPLIT_COMMUNICATION)), serial)
    # add serial.c
    SPLIT_SRC += $(QUANTUM_DIR)/split_common/serial.c
    # set SPLIT_COMMUNICATION_SERIAL for matrix.c and split_util.c
    OPT_DEFS += -DSPLIT_COMMUNICATION_SERIAL
  else
    ifeq ($(strip $(SPLIT_COMMUNICATION)), i2c)
      # set SPLIT_COMMUNICATION_I2C for matrix.c and split_util.c
      OPT_DEFS += -DSPLIT_COMMUNICATION_I2C
      USE_I2C = yes
    else
      $(error invalid SPLIT_COMMUNICATION value)
    endif
  endif

  ifeq ($(strip $(OLED_ENABLE)), yes)
    SPLIT_SRC += ssd1306.c
    # set SSD1306OLED for split_util.c
    OPT_DEFS += -DSSD1306OLED
    USE_I2C = yes
  endif

  # is there any module using I2C?
  ifeq ($(strip $(USE_I2C)), yes)
    # add i2c.c
    SPLIT_SRC += $(QUANTUM_DIR)/split_common/i2c.c
    # set USE_AVR_ATmega32U4_I2C for seria.c(for conflict check)
    OPT_DEFS += -DUSE_AVR_ATmega32U4_I2C
  endif

  ifdef SPLIT_RULES_VERBOSE
    $(info *** split keyboarad configuration ***)
    $(info  SRC += $(SPLIT_SRC))
    $(info  USE_I2C = $(USE_I2C))
    $(info  SPLIT_COMMUNICATION = $(SPLIT_COMMUNICATION))
    $(info  OPT_DEFS = $(OPT_DEFS))
  endif

  QUANTUM_SRC += $(SPLIT_SRC)
endif
```

drashna, 2018-11-18:
I don't think all of the feature code is appropriate here.   A lot of this should be relegated to defines, and not feature rules. 
But that's more of semantics, though. 

Additionally, for the boards using i2c for something other than split communication, it may be better (long term) to use the main i2c driver, and not the split code (though, the split code should see about using the main driver too, eventually).
You can see the i2c drivers here: https://github.com/qmk/qmk_firmware/tree/master/drivers/avr

This would allow for much more advanced configurations, and remove some of the "mess" that exists right now. 

Additionally, in theory, for the boards like the helix, because i2c uses addresses, it theory you could use i2c for both split communication and for communicating with things like the OLED displays (and both of them, actually).  However, I do understand that this is much more complicated to implement, and cleanly.

mtei, 2018-11-18:
I think that it is best that the objects of drivers is contained in the library file as follows.

In keyboards/KEYBOARD/config.h
```c
#define SPLIT_COMMUNICATION_SERIAL
// or #define SPLIT_COMMUNICATION_I2C
```

In matrix.c and split_util.c
```c
#if defined(SPLIT_COMMUNICATION_I2C)
 // using I2C code
#elif defined(SPLIT_COMMUNICATION_SERIAL)
 // using serial code
#endif
```

compile and link sequence
```
$ avr-gcc -c split_common/serial.c
$ avr-gcc -c split_common/i2c.c  # or drviers/avr/i2c_master.c ??
$ avr-ar cvr libdrivers.a serial.o i2c.o
...
$ avr-gcc -c <each .c files>
...
$ avr-gcc -o keyboard.elf $(OBJS) -ldrivers.a # automatically select serial.o, i2c.o
```

But I didn't know how to add library generation rules to the Qmk_firmware makefile. As the second best method, I thought the above qmk_firmware/common_features.mk idea.

mtei, 2018-12-07:
I posted #4522. This is a concrete solution I thought about comments https://github.com/qmk/qmk_firmware/issues/4254#issuecomment-439307124 and https://github.com/qmk/qmk_firmware/issues/4254#issuecomment-439717577.

mtei, 2018-12-12:
> Additionally, there is the `i2c_master.c` "driver" available. Using that may be a good idea for the split keyboards in general. But may be much better for the OLED screen support, in general. As this would cause less issues with the split stuff.
> 
> rgb matrices, and qwiic devices use i2c, and use the /drivers/*/i2c_master.c file.

Currently, compilation of `i2c.c` and `serial.c` under `quantum/split_common/` is exclusively switched depending on whether there is a `USE_I2C` macro or not.

I think it's a good idea to change to use a common driver rather than using its own `i2c.c` under `quantum/split_common/`.  If so, we should stop using the `USE_I2C` macro as a compilation switch.

PullRequest #4522 provides a method to compile both i2c.c and serial.c and link only what is needed at link time.

mtei, 2018-12-13:
> Also, this is unrelated, but the OLED screen code in the keymap... you could use `biton32(layer_state)` to handle the layer codes, that way, you don't have to use the actual mask.
> In fact, I'm doing that on the crkbd, and verify that it works well:

Thank you for telling me `biton32()`. It was good to know.

That code was copied from Helix keymap 'five_rows'.  And in 'five_rows', I wanted to display all the active layers.  Because 'five_rows' has a few layers that are almost transparent.
So `biton32()` can not be used in this case.

[image]

drashna, 2018-12-14:
Ah, yeah, if you wanted to display all of the layers, then that wouldn't work.  But I'm glad that I mentioned it, regardless.

mtei, 2018-12-15:
I am currently working on bringing helix-serial.c to quantum/split_common.

 It is known that keyboard MiniAxe can not be compiled. 

https://github.com/mtei/qmk_firmware/commits/replace_serial_c_of_quantum_split_common

mtei, 2018-12-15:
It is almost completed, and I also confirm that it works. 

However, miniaxe has been remodeled by copying matrix.c from `split_common/`, so it is stranded from the change of `splir_common/matrix.c` this time.

mtei, 2018-12-16:
I contacted @ka2hiro, creator of miniaxe, and I got permission to change keyboards/miniaxe/matrix.c.

TheOneTrueTrench, 2019-02-04:
I've got an Iris and a Quefrency, I'm entirely willing to resolder Proton-C's on them and work on testing them.

TheOneTrueTrench, 2019-02-06:
Can you link me to your branch? I've been wanting to understand how it works.

Gimly, 2019-02-23:
I would like to help adding support for more than 8 columns since I have that need for a keyboard I have built that is a split TKL and has 11 columns (on the right side, 8 on the left).

I've seen some TODOs in the transport.c code that talks about using pack/unpack, is this still the way to go and what are those methods? Do someone have a link explaining them?

pelrun, 2019-02-24:
Serial has supported >8 columns since before I started work on this, it was only i2c that was restricted.

Gimly, 2019-02-24:
@pelrun I'm using i2c, so is it still restricted to <= 8 columns in i2c, and if so what needs to be done to fix that?

Lenbok, 2019-02-24:
@Gimly You can help test  #4974 which aims to remove the column restriction for i2c splits.

drashna, 2019-02-24:
@Gimly The current issue is that the i2c code in #4974 does not work.  I'm not sure why, as the code looks okay.

That PR fixes the column limit, already.  But the i2c code needs work and testing.

Once that's done, we can merge it and be significantly closer to ARM support as well.

pelrun, 2019-02-24:
> The current issue is that the i2c code in #4974 does not work. 

It's been fixed for a couple of days now :)

Gimly, 2019-02-24:
Great, I'll do some tests tomorrow with #4974 code. I'll update with my findings. Thanks!

yurikhan, 2019-02-27:
+1 for another method of master detection. I am using an Arduino clone (CNT-013) that shorts VUSB to +5V so the usual method will not work.

`vitamins_included` has a different solution: power on as slave, become master when assigned a USB address, or continue as slave when contacted by a master over I2C.

XScorpion2, 2019-03-02:
So I did a file based analysis on the i2c.h/c versions in use in qmk and the results can be found here:
https://drive.google.com/open?id=1ySZlhbUucAQoQBnpiZ8rYX3LtNFvkdnP
There are basically 44 copies, that boil down to 3 unique versions: split common, split common with oled functions, and the Luiz Ribeiro's version. If you open i2c_analysis.txt in that drive folder, it goes into more details.

drashna, 2019-03-20:
Updated the main post, with additional information, and ARM stuff

DeepwaterCreations, 2019-06-03:
Hey, is there a good way for a random coder such as myself to help out with this issue? I'm itching to get my Proton-C + Viterbi build underway. :)

drashna, 2019-06-03:
absolutely.   

RIght now, the remaining issues are: 
* No i2c_slave.c for ARM
* Serial is hard coded for AVR.  
   * It would need to be generalized, or a separate one written for ARM, and the existing one moved to /drivers/avr/
* i2c_master for AVR is not properly licensed, and needs to be replaced with a GPL Compliant version. 

From there, I think everything is ready for ARM.  But we won't know for certain until those are fixed up. ,

@pelrun has been the one spearheading it, but I'm not sure where he is, at it, right now.   
@mtei has also been very active with the split keyboard code, and may be able to help. 

If you want to contribute and help here, and have the experience, then it may be worth heading to discord and chatting up pelrun.  and others.  And chatting on the progress.

kyleterry, 2019-06-12:
What is the status of this issue?

I've been working on revisions of my own split board called the Spacetime keyboard that's similar to the Corne and the helix as far as the circuit goes. My goal is to not move any of the serial and split code from those keyboards over to the spacetime board and to use split common and the avr i2c code in qmk core.

The keyboard has 7 columns and 4 rows on each half, an oled on each half, 50 rgb led matrix (25 on each half) and uses serial over TRRS on D2. The oleds are attached to scl and sda for i2c and the leds are attached to D3.

A lot of stuff works right away without a lot of configuration. RGB backlight with animations work, although there is some serious power draw issues, so I have to limit the brightness pretty significantly, otherwise the slave half locks up when the controller doesn't have enough power. This might be a hardware design problem I have, but I think it's a known thing that can happen with a lot of leds.

RGB matrix does not work with the core serial split code. the slave half doesn't function right if I try to use the matrix feature. This might be my matrix code, but I would like to confirm if this is a known thing.

The oleds do not work with the core serial split and i2c code. I think this is known as well (it looks like there were a lot of changes made to serial and i2c code out-of-band in the crkbd and helix boards), but I would like to confirm that is a thing as well.

So with all that said, I have a board I can experiment with to get some of these features moved over, I just need to know what are known broken things that aren't being worked on that I can fix. Since my board functions almost exactly like those two boards, I have a clean slate to work with and can work on getting things up and running.

drashna, 2019-06-12:
@kyleberry it's mostly there.  

But your issue is more with how the split code works in general.  #5998 seeks to address that issue, actually.  And I've tested it with the Corne using #6001 as well, and can verify it works.

cmpulsip, 2019-06-12:
Hello @drashna,
I've been following the progress of this issue very closely as I'd like to build a split keyboard using two ARM-based microcontrollers(STM32) one day. I'm new to building a firmware, but would like to get more familiar and help in any way I can. I see that you replied to @DeepwaterCreations above about the current state of this issue. Does that list still stand, or is there another aspect in which I might offer a small contribution?

Thanks.

drashna, 2019-06-12:
The list is up to date. 

Serial and i2c_slave support need to be added for ARM, at this point.

paraqles, 2020-03-02:
Anything new about this?

I'm unfortunately not so well versed with C. And especially not with ARM. Otherwise I would like to help. Are there any starting points to get into position to help?

drashna, 2020-04-20:
It's mostly there.  

There are a few PRs open that are relevant, and it may be worth taking a look at. 

Jumping onto discord may be the best best for talking about it

ClaudiaLapalme, 2021-05-28:
What's the status on this? Can I help?

tzarc, 2021-05-29:
Should be pretty much complete at this point. Will have a look through things this weekend and see if there's anything else outstanding.

drashna, 2021-07-25:
Closing this, because, yeah, this is pretty much complete.

Only thing missing is i2c_slave for chibiOS, but that's ... a can of worms. 

Also, helix should be ported over

closed 2021-07-25 by drashna

mtei, 2021-09-01:
### Report about Helix keyboad scan rate

With the improvements made by #11930, the performance of split_common has caught up with the local implementation of Helix (rev2).

#### before 0.14.0
| use split_common |   oled   | rgblight back | scan rate |
|------------------|----------|---------------|-----------|
| no (local)       |  disable | disable       | 1631      |
| no (local)       |  enable  | enable        | 1515      |
| yes              |  disable | disable       | 880       |
| yes              |  enable  | enable        | 829       |

#### after 0.14.0
| use split_common |   oled   | rgblight back | scan rate |
|------------------|----------|---------------|-----------|
| no (local)       |  disable | disable       | 1627      |
| no (local)       |  enable  | enable        | 1513      |
| yes              |  disable | disable       | 1555      |
| yes              |  enable  | enable        | 1360      |

So, Helix keymaps that have been switched to using oled_driver.c in the OLED code can be switched to use split_common.