qmk/qmk_firmware #4255: Planck rev6 not working after #3914

warptozero, 2018-10-27:
At first I thought it was because of my keymap, but then I remembered I did a `git pull` earlier and after some checkouts narrowed it down to #3914. I can reproduce it with a fresh `git clone --recurse-submodules`.

To flash I run `make planck/rev6:default_mod:dfu-util` and the firmware compiles and flashes without any errors. But after sounding the startup sound all the keys are unresponsive.

Let me know if I'm doing something wrong here and what you need me to do for debugging.

drashna, 2018-10-27:
On an off chance, could you put `eeconfig_enable()` and `eeconfig_init()` in your `matrix_init_user` function and see if that fixes the issue?

warptozero, 2018-10-27:
Well this is weird, I can't seem to reproduce this anymore, even though I could do so reliably a few hours ago and as far as I can tell nothing else changed.

warptozero, 2018-10-27:
However, I did put those 2 functions in before flashing the first time, and after that commenting them out again to recheck. And when I look at the source I see they reset the EEPROM. So could it be that running them once actually fixed the problem?

drashna, 2018-10-27:
Yeah, you'd only need to run those once, and then you'd want to remove them from your code.  

And if that fixed them, then I have a good idea of what happened, and it's my fault.  I mistakenly changed some of the eeprom initialization handling, and that may have caused the issue.   I discovered that issue with messing with the bootmagic lite code, and that PR does fix the issue, as well.

warptozero, 2018-10-28:
Well that's good news then. Thanks for jumping on it so fast.

drashna, 2018-10-28:
Well, I found some eeprom handling issues when messing with bootmagic lite (which resets the eeprom by default, too).  

And Bootmagic Lite has been merged, so if that was the cause, it shouldn't happen anymore. 

(Also, I may live and breath QMK anymore :D )