qmk/qmk_firmware #4257: dz 60 keyplacement issue

blackbaddl, 2018-10-27:
Hi everyone,
i have build my first keyboard with a dz 60 and enyoing it so far.... but there is one mayor problem. living in germany and have been able to type with 10 fingers without any looking for at least 6 years. tried to get used to the switched placement of "z" and "y" but i cant ... tried to switch it in qmk firmware flashed it multiple times but it does not work. all changes i make are working perfectly but z and y are not changing their position at all. i tried several source to make my hex file with but always the same(https://config.qmk.fm/ or http://qmkeyboard.cn/). what could my problem be... quite frustrated at the moment. and there is a second question i did not find an answer for. when completely going with C for keymap programming what should i use to get my flashable hex file. i am aware of that there are several more file which have to be included like rules.mk and so on.

thanks 
best regards

blackbaddl, 2018-10-27:
have an ugly software fix right now but there must be a qmk solution. so far i can deal with it but wanna fix this in the near future

drashna, 2018-10-27:
Well, first, you're OS keyboard layout has an influence here.  That may not be set correctly.

Secondly, you may want to include this:
https://github.com/qmk/qmk_firmware/blob/master/quantum/keymap_extras/keymap_german.h

And use those keycodes instead.  That may fix your issues.

You would need to add `#include "keymap_german.h"` to your keymap.c file.

closed 2019-11-21 by drashna