qmk/qmk_firmware #4271: undefined reference to `console_printf' when enabling CONSOLE_ENABLE for Massdrop Alt

rainer-ih, 2018-10-28:
When enabling CONSOLE_ENABLE for the Massdrop Alt and running `make massdrop/alt:default`, the following errors occur:

```
 | .build/obj_massdrop_alt_default/matrix.o: In function `matrix_print':
 | qmk_firmware-master/keyboards/massdrop/alt/matrix.c:181: undefined reference to `console_printf'
 | .build/obj_massdrop_alt_default/keyboards/massdrop/alt/keymaps/default/keymap.o: In function `process_record_user':
 | qmk_firmware-master/keyboards/massdrop/alt/keymaps/default/keymap.c:84: undefined reference to `console_printf'
 | qmk_firmware-master/keyboards/massdrop/alt/keymaps/default/keymap.c:86: undefined reference to `console_printf'
 | qmk_firmware-master/keyboards/massdrop/alt/keymaps/default/keymap.c:170: undefined reference to `console_printf'
 | qmk_firmware-master/keyboards/massdrop/alt/keymaps/default/keymap.c:170: undefined reference to `console_printf'
 | .build/obj_massdrop_alt_default/keyboards/massdrop/alt/keymaps/default/keymap.o:qmk_firmware-master/keyboards/massdrop/alt/keymaps/default/keymap.c:175: more undefined references to `console_printf' follow
 | collect2: error: ld returned 1 exit status
 |
make[1]: *** [.build/massdrop_alt_default.elf] Error 1
make: *** [massdrop/alt:default] Error 1
```

drashna, 2018-10-28:
Tagging @patrickmt, since you may have the best idea on the issue, and how to fix.

@rainer-ih thanks for opening this on github!

patrickmt, 2018-10-29:
Not sure how this bug made it through. Thanks for submitting it and @drashna for bringing it to my attention!

rainer-ih, 2018-10-29:
> Not sure how this bug made it through. Thanks for submitting it and @drashna for bringing it to my attention!

Hey @patrickmt, is there a change I can make locally to fix this or is it a complex fix? Thanks very much for looking!

patrickmt, 2018-10-29:
@rainer-ih You can drill through the referenced pull request to see the changes. Ultimately you would end up here: https://github.com/qmk/qmk_firmware/pull/4277/files. You can see changes on lines 18 and 66 of the file tmk_core/common/arm_atsam/printf.c, where tests on define CONSOLE_PRINT is changed to tests on define CONSOLE_ENABLE

rainer-ih, 2018-10-29:
> @rainer-ih You can drill through the referenced pull request to see the changes. Ultimately you would end up here: https://github.com/qmk/qmk_firmware/pull/4277/files. You can see changes on lines [...]
I'll give that a try, thanks a lot!

drashna, 2018-10-29:
@rainer-ih I merged this PR, so if you update your repo, you should have this fix now. 

If you need help on updating the repo, @noroadsleft has an awesome guide that he wrote on how to do that, actually:
https://docs.qmk.fm/#/newbs_best_practices

rainer-ih, 2018-10-30:
Thanks guys, really appreciate it 👍🏼

closed 2018-10-30 by drashna