qmk/qmk_firmware #4275: Random keycodes output for one specific board

colemarkham, 2018-10-29:
As discussed on Discord, getting random garbled keycodes only with the new keyboard I'm working on. Other keyboards compile and run fine with the same codebase and virtually identical code. I was ultimately able to make it work by adding more dummy layers. A total of 8 are required, any less and it doesn't work. Debugging statements on the layer_state variable indicate it is set to layer 0. Fix that ultimately made it work: https://github.com/colemarkham/qmk_firmware/commit/3d3f1f41ab360fb9335486839deca9392729189c

The behavior is consistent with attempting to access a layer that is undefined. To verify, I added a keycode to indicate which layer is being used (https://github.com/colemarkham/qmk_firmware/commit/d88d3e777343b3de88c7b82602048947f8884366). This shows that it is indeed layer 7 even though the layer_state variable specifies otherwise. Attempts to clear the eeprom have not made any difference.

Any thoughts on how to debug this further?

jackhumbert, 2018-10-29:
If you fill each layer with keycodes besides `KC_TRNS`, what layer does it think it's on?

colemarkham, 2018-10-29:
That's what I did in the 2nd commit. It thinks it's on 7, the last one.

colemarkham, 2018-10-31:
Fixed by resetting the EEPROM. Was trying to use QMK Toolbox which doesn't support atmega32u2 and doesn't inform the user of that fact: https://github.com/qmk/qmk_toolbox/issues/56

closed 2018-10-31 by colemarkham