qmk/qmk_firmware #4282: Planck rev6 "TERMINAL_ENABLE = yes" breaks build

chax, 2018-10-30:
If you put `TERMINAL_ENABLE = yes` in rules.mk for planck rev6, build breaks with following error:

```
$ make planck/rev6:default
QMK Firmware 0.6.159
Making planck/rev6 with keymap default

arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/planck/rev6/matrix.c                                                           [OK]
Compiling: quantum/audio/muse.c                                                                     [OK]
Compiling: keyboards/planck/planck.c                                                                [OK]
Compiling: keyboards/planck/rev6/rev6.c                                                             [OK]
Compiling: keyboards/planck/keymaps/default/keymap.c                                                [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/keymap_common.c                                                                  [OK]
Compiling: quantum/keycode_config.c                                                                 [OK]
Compiling: quantum/process_keycode/process_audio.c                                                  [OK]
Compiling: quantum/process_keycode/process_clicky.c                                                 [OK]
Compiling: quantum/audio/audio_arm.c                                                                [OK]
Compiling: quantum/audio/voices.c                                                                   [OK]
Compiling: quantum/audio/luts.c                                                                     [OK]
Compiling: quantum/process_keycode/process_music.c                                                  [OK]
Compiling: quantum/process_keycode/process_terminal.c                                              In file included from tmk_core/common/print.h:76:0,
                 from tmk_core/common/debug.h:22,
                 from quantum/keymap.h:35,
                 from quantum/quantum.h:29,
                 from quantum/process_keycode/process_terminal.h:20,
                 from quantum/process_keycode/process_terminal.c:17:
tmk_core/common/chibios/printf.h:108:16: error: conflicting types for 'tfp_printf'
 #define printf tfp_printf
                ^
tmk_core/common/chibios/printf.h:103:6: note: previous declaration of 'tfp_printf' was here
 void tfp_printf(char *fmt, ...);
      ^~~~~~~~~~
tmk_core/common/chibios/printf.h:109:17: error: conflicting types for 'tfp_sprintf'
 #define sprintf tfp_sprintf
                 ^
tmk_core/common/chibios/printf.h:104:6: note: previous declaration of 'tfp_sprintf' was here
 void tfp_sprintf(char* s,char *fmt, ...);
      ^~~~~~~~~~~
 [ERRORS]
 | 
 | 
 | 
make[1]: *** [tmk_core/rules.mk:359: .build/obj_planck_rev6_default/quantum/process_keycode/process_terminal.o] Error 1
Make finished with errors
make: *** [Makefile:544: planck/rev6:default] Error 1

```

chax, 2018-10-30:
managed to build it finally by adding `#define USER_PRINT` in config.h 
and `TMK_COMMON_SRC += $(CHIBIOS)/os/various/syscalls.c` in rules.mk

maybe someone with more experience can make fix and PR for this

drashna, 2018-10-30:
could you run `git diff `, and copy the output here?

chax, 2018-10-30:
oh and i also commented out this line
https://github.com/qmk/qmk_firmware/blob/master/tmk_core/common/print.h#L76

these are all the changes i did to make this work, but i know this is not the proper way to fix it.

chax, 2018-10-30:
Yeah i know how to contribute, it's just that i don't have enough experience to make **proper** fix.

chax, 2018-10-30:
With these changes i fixed it for myself but i probably broke it for someone else.

drashna, 2018-10-30:
Okay.

As for a proper fix, as long as it works, and doesn't break anything else, it's proper enough.

But you could wrap then in ifndef's for TERMINAL_ENABLE, and that should be good enough for now.

chax, 2018-10-30:
And in which rules.mk (or some other makefile) should i put this: 
`TMK_COMMON_SRC += $(CHIBIOS)/os/various/syscalls.c`

chax, 2018-10-30:
This works for me
Now i can just add or remove `TERMINAL_ENABLE = yes` in rules.mk and builds in both cases.

```
$ git diff                                                                                                                                                                                                                                                                                                                                                                                                     18.1s  30.10.2018 03:00:04
diff --git a/keyboards/planck/config.h b/keyboards/planck/config.h
index 452a99d74..b071e77e2 100644
--- a/keyboards/planck/config.h
+++ b/keyboards/planck/config.h
@@ -86,4 +86,8 @@ along with this program.  If not, see <http://www.gnu.org/licenses/>.
 //#define NO_ACTION_MACRO
 //#define NO_ACTION_FUNCTION
 
+#ifdef TERMINAL_ENABLE
+  #define USER_PRINT
+#endif
+
 #endif
diff --git a/tmk_core/common.mk b/tmk_core/common.mk
index 3844b13d4..a166c0676 100644
--- a/tmk_core/common.mk
+++ b/tmk_core/common.mk
@@ -46,6 +46,8 @@ ifeq ($(PLATFORM),CHIBIOS)
 endif
   ifeq ($(strip $(AUTO_SHIFT_ENABLE)), yes)
     TMK_COMMON_SRC += $(CHIBIOS)/os/various/syscalls.c
+  else ifeq($(strip $(TERMINAL_ENABLE)), yes)
+    TMK_COMMON_SRC += $(CHIBIOS)/os/various/syscalls.c
   endif
 endif
 
diff --git a/tmk_core/common/print.h b/tmk_core/common/print.h
index d94527657..06c6cbd7f 100644
--- a/tmk_core/common/print.h
+++ b/tmk_core/common/print.h
@@ -73,7 +73,9 @@ void print_set_sendchar(int8_t (*print_sendchar_func)(uint8_t));
 
 #elif defined(PROTOCOL_CHIBIOS) /* PROTOCOL_CHIBIOS */
 
+#ifndef TERMINAL_ENABLE
 #  include "chibios/printf.h"
+#endif
 
 #  ifdef USER_PRINT /* USER_PRINT */

```

PR #4299 (2018-11-06): Fix Terminal feature on ChibiOS
Fixes #4282
closed 2018-11-06 by jackhumbert