qmk/qmk_firmware #4302: Advanced Combos

NotGate, 2018-10-31:
QMK does not currently support combos that share a common key. A reimplementation of Combo that supports all possible combinations of keys would need to be robust enough to support a large number of possibilities without overlaps causing any detriment to performance. This can most likely be done by calculating the 'value' of the matrix and indexing an array/calling a function based off that value, whether keys are shared or not shouldn't matter. 

Example: Using a 2x3 set of keys to type all 26 letters of the alphabet. 

[image]

This is how I would think to implement combinations on a small scale. A larger grid would have numbers too big for a 16 bit or 32 bit integer to hold but all that matters is that each combo is uniquely identifiable just based off the state of the matrix. I don't know too much about the specifics of Combos current implementation or if doing it this way would run into difficulties. The actual usage of Combo should look the same, since this calculation and the indexing should just be internal. 

Getting this working is a priority of mine so any help/advice is greatly appreciated. I'll be working to try and get it working locally on my planck/rev6.

rsheldiii, 2019-04-09:
With https://github.com/qmk/qmk_firmware/pull/2561 merged, this is now possible, with one caveat. If one combo is a subset of another combo, the first combo always executes when the second is pressed. That is to say, if a+b = q and a+b+d = w, when pressing a+b+d `qw` is the output.

This won't make the example above possible, as a 'value' of 111 would trigger the combos for values 110 and 100. Given 6 letters though, `6 choose 3 = 20`, and the base 6 letters can be typed for their values. You can indeed type the entire alphabet with 6 letters, you just have to do it with 6 1-letter keys and 20 3-letter combos.

alexspeller, 2019-05-31:
#6029 fixes this, it does still allow combos as modifiers even if the flag is turned on.

closed 2019-12-21 by stale[bot]

danielo515, 2020-08-04:
I was about to open an issue to report this and I found this issue. Is it then a known and accepted limitation? How do steno keyboards resolve this problem?
Is this documented on the combo docs?

noahfrederick, 2020-08-04:
> Is it then a known and accepted limitation?

The original issue was about combos sharing keys at all—this issue is fixed so that combos may share keys (e.g., `ab` and `bc`). But the secondary issue is that combos may not be subsets of other combos (e.g., `ab` and `abc`)—this is a limitation that a couple of people have tried to fix but no one has followed through yet.

> How do steno keyboards resolve this problem?

Steno keyboards don't use the combo feature. Mostly they use a steno protocol which doesn't relate to this functionality. There are also some keyboards that implement their own combo engine, such as those of g Heavy Industries.

I believe this was the most promising work towards getting this limitation removed: https://github.com/qmk/qmk_firmware/pull/6029

danielo515, 2020-08-04:
Ok, true, my problem is with combos being subsets of other combos.
Thanks for the reference links.