qmk/qmk_firmware #4306: AVR XMEGA MCUs not properly supported

drashna, 2018-10-31:
https://www.reddit.com/r/olkb/comments/9sz2of/qmk_xmega_support/

The ARCH, F_CPU and F_USB values have to be changed to get this to start compiling, but it errors out on the suspend stuff.

```
Compiling: tmk_core/common/avr/suspend.c                                                           In file included from c:\msys64\home\drashna\qmk_utils\avr8-gnu-toolchain\avr\include\avr\io.h:99:0,
                 from c:\msys64\home\drashna\qmk_utils\avr8-gnu-toolchain\avr\include\avr\sleep.h:38,
                 from tmk_core/common/avr/suspend.c:2:
tmk_core/common/avr/suspend.c: In function 'power_down':
tmk_core/common/avr/suspend.c:38:14: error: 'WDCE' undeclared (first use in this function)
     "r" (_BV(_WD_CHANGE_BIT) | _BV(WDE)), \
              ^
tmk_core/common/avr/suspend.c:106:5: note: in expansion of macro 'wdt_intr_enable'
     wdt_intr_enable(wdto);
     ^
tmk_core/common/avr/suspend.c:38:14: note: each undeclared identifier is reported only once for each function it appears in
     "r" (_BV(_WD_CHANGE_BIT) | _BV(WDE)), \
              ^
tmk_core/common/avr/suspend.c:106:5: note: in expansion of macro 'wdt_intr_enable'
     wdt_intr_enable(wdto);
     ^
tmk_core/common/avr/suspend.c:38:36: error: 'WDE' undeclared (first use in this function)
     "r" (_BV(_WD_CHANGE_BIT) | _BV(WDE)), \
                                    ^
tmk_core/common/avr/suspend.c:106:5: note: in expansion of macro 'wdt_intr_enable'
     wdt_intr_enable(wdto);
     ^
tmk_core/common/avr/suspend.c:40:13: error: 'WDIE' undeclared (first use in this function)
         _BV(WDIE) | (value & 0x07)) ) \
             ^
tmk_core/common/avr/suspend.c:106:5: note: in expansion of macro 'wdt_intr_enable'
     wdt_intr_enable(wdto);
     ^
In file included from tmk_core/common/avr/suspend.c:4:0:
tmk_core/common/avr/suspend.c: In function 'WDT_vect':
tmk_core/common/avr/suspend.c:211:5: error: 'WDT_vect' appears to be a misspelled signal handler, missing __vector prefix [-Werror=misspelled-isr]
 ISR(WDT_vect)
     ^
cc1.exe: all warnings being treated as errors
 [ERRORS]
 |
 |
 |
make[1]: *** [tmk_core/rules.mk:359: .build/obj_xmega_default/common/avr/suspend.o] Error 1
```

It looks like there is a lot of changes here to support the AVR XMEGA MCUs.

ResoluteFury, 2018-10-31:
Also errors on timers.

`Compiling:` tmk_core/common/avr/timer.c                                                             tmk_core/common/avr/timer.c: In function ‘timer_init’:
tmk_core/common/avr/timer.c:52:5: error: ‘TCCR0A’ undeclared (first use in this function)
     TCCR0A = 0x02;
     ^
tmk_core/common/avr/timer.c:52:5: note: each undeclared identifier is reported only once for each function it appears intmk_core/common/avr/timer.c:54:5: error: ‘TCCR0B’ undeclared (first use in this function)
     TCCR0B = prescaler;
     ^
tmk_core/common/avr/timer.c:56:5: error: ‘OCR0A’ undeclared (first use in this function)
     OCR0A = TIMER_RAW_TOP;
     ^
tmk_core/common/avr/timer.c:57:5: error: ‘TIMSK0’ undeclared (first use in this function)
     TIMSK0 = (1<<OCIE0A);
     ^
tmk_core/common/avr/timer.c:57:18: error: ‘OCIE0A’ undeclared (first use in this function)
     TIMSK0 = (1<<OCIE0A);
                  ^
In file included from tmk_core/common/avr/timer.c:19:0:
tmk_core/common/avr/timer.c: In function ‘TIMER0_COMPA_vect’:
tmk_core/common/avr/timer.c:145:32: error: ‘TIMER0_COMPA_vect’ appears to be a misspelled interrupt handler, missing __vector prefix [-Werror=misspelled-isr]
 #define TIMER_INTERRUPT_VECTOR TIMER0_COMPA_vect
                                ^
tmk_core/common/avr/timer.c:149:5: note: in expansion of macro ‘TIMER_INTERRUPT_VECTOR’
 ISR(TIMER_INTERRUPT_VECTOR, ISR_NOBLOCK)
     ^
cc1: all warnings being treated as errors
 [ERRORS]
 |
 |
 |`

closed 2021-04-18 by drashna

eriqadams, 2021-12-01:
Can I use this atxmega32a4u in QMK now ?