qmk/qmk_firmware #4321: Combos not working on ARM boards (planck/rev6)

NotGate, 2018-11-02:
On a planck/rev6, using only the default keymap and combo examples, combos do not work as expected.

A combo of a+b->c will produce something like 'cab' or 'caaaaaab' depending on whether COMBO_ALLOW_ACTION_KEYS is defined or not, respectively. Trying other keymaps, combo configurations, and defines doesn't seem to help. 

The attached file is the default planck layout using the first example listed at https://docs.qmk.fm/#/feature_combo

[keymap.txt](https://github.com/qmk/qmk_firmware/files/2540956/keymap.txt)

kpgalligan, 2019-01-21:
I can confirm this is happening on at least the Massdrop Ctrl. This is my first keyboard hacking attempt, so I don't have much other info to share. Still getting my bearings.

onetom, 2019-01-24:
This might be partly caused by https://github.com/qmk/qmk_firmware/issues/4424

equalunique, 2019-02-27:
I noticed the same thing on my planck rev6 today. Tapping / combo terms ranging from 20 to 1000 do not affect the recurrence of the issue. The layout I am using almost identical to the default layout + the example AB_ESC and JK_TAB combos from the QMK docs.

rsheldiii, 2019-05-03:
now that #4424 is closed, can anyone check if this is still true?

ibnuda, 2021-09-19:
i am also experiencing this issue. though with the following context:
1. firmware: https://github.com/ibnuda/qmk_firmware/tree/rickyrickrarded/keyboards/neva4get
2. set `COMBO_ENABLE = yes`
3. compile and flash.
4. will output the following lines in the `dmesg` output. and leave the keyboard not working. not even sending keycodes at all when pins, say, R1 and C1 are shorted 
```
[ 2529.211005] hid-generic 0003:1209:6971.0023: input,hidraw1: USB HID v1.11 Keyboard [Ibnu D. Aji Neva4Get] on usb-0000:00:14.0-1/input0
[ 2529.211642] hid-generic 0003:1209:6971.0024: hiddev96,hidraw2: USB HID v1.11 Device [Ibnu D. Aji Neva4Get] on usb-0000:00:14.0-1/input1
```
5. set `COMBO_ENABLE =  no`
6. will output the following lines and the keyboard will work just fine.
```
[ 3242.113330] hid-generic 0003:1209:6971.0029: input,hidraw1: USB HID v1.11 Keyboard [Ibnu D. Aji Neva4Get] on usb-0000:00:14.0-1/input0
```

closed 2022-08-28 by zvecr