qmk/qmk_firmware #4327: Viterbi w/ Elite-C: 3 columns opposite master side don't work

mcmillion, 2018-11-02:
I just finished a build of a new Viterbi with the new Elite-C controllers. When testing it with my existing Viterbi layout, I noticed that the last 3 columns don't respond. What's interesting is that if you plug into the other controller then this flips. For example:

* When plugged into the left side, the far right 3 columns don't work.
* When plugged into the right side, the far left 3 columns don't work.

I've also played with the config settings for the hands and have seen that the keymap does reverse as it's supposed to, so I'm leaning towards this being a firmware thing and not something electrical.

Any ideas? Do I need to go I2C?

drashna, 2018-11-02:
Are you compiling from source? 

If so, you need to update your repo.  

The issue is that these columns are the JTAG pins, as well, and it's not getting disabled properly. 

There is a fix that is out for this already, but it's a day or three old.

mcmillion, 2018-11-02:
Bingo! I had done some digging and figured it was related to the JTAG stuff, but couldn't figure it out. Turns out my repo was a day older than when the fix went in, so merging in upstream changes and recompiling and flashing both sides worked like a charm. Thanks!

closed 2018-11-02 by mcmillion

drashna, 2018-11-02:
Awesome!!

Glad to hear it. :)