qmk/qmk_firmware #4332: Fade not working when key pressed

Rainonmepls, 2018-11-03:
I flash the program but all of the fade effects are not working

drashna, 2018-11-03:
What keyboard is this?

Rainonmepls, 2018-11-03:
I am using massdrop ctrl keyboard

drashna, 2018-11-03:
Okay, that's what I thought.  I'm tagging @patrickmt here, as he is more familiar with the Massdrop rgb matrix code than I.

Rainonmepls, 2018-11-03:
do u have the same problem as me ?

drashna, 2018-11-03:
I don't have a CTRL or ALT, which is why I'm not very familiar with the code for it, at all.   It's not the "RGB Matrix" code from the QMK feature, but custom stuff.

DarkMio, 2018-11-03:
Describe the effect you are interested in and the steps you took to do it.

Do all LEDs operate normally in other modes? What do you do to enable said effect on your keyboard?

Rainonmepls, 2018-11-03:
I wan the effects to be like this video https://www.youtube.com/watch?v=ntZn0AcSwt8
I just flash the program that was on the repo

DarkMio, 2018-11-03:
That is not on this repository. This is part of my fork on my branch of the repository and is not part of the normal firmware.

You can checkout my repository with a branch called massdrop_mio or so and compile with ``massdrop/CTRL:darkmio`` or something like that - this should give you the scrolling fade. Otherwise, the video linked has a repository linked that has that written.

DarkMio, 2018-11-03:
There is no firmware compiled in the video description. It is true that I do only have the scrolling effect, while @matsprehn has a fading only effect.

You likely have compiled my repo when you wanted to compile the other repository.

Rainonmepls, 2018-11-03:
i can't seem to find your repo and I uploaded his file but he also have the scrolling fade effect

closed 2019-11-21 by drashna