qmk/qmk_firmware #4334: OSM toogling doesn't work, only work in the bugged scenario when other OSM keys are pressed

mnpqraven, 2018-11-03:
* Setup
Alt, Shift, Ctrl, basically all modifier keys are used with OSM()
in my `config.h`:
````
#define ONESHOT_TAP_TOOGLE 2
#undef ONESHOT_TIMEOUT
````
* Problem
Toggling the OSM lock by tapping the keys doesn't work(example when i double tap shift, the first tap doesn't appear on the key tester program, which is working as intended and then the 2nd tap only taps the shift and not hold it), only time it worked was when i deliberately try to replicate the bug in #3963 which ironically also has its own problems
[Here is my keymap folder](https://github.com/mnpqraven/qmk_firmware/tree/master/keyboards/facew/keymaps/othi) if anyone's willing to take a look

drashna, 2018-11-03:
To clarify, the second tap would lock it enabled.  

Specifically it does this by basically "pressing and holding" down the modifier key. Shift in this case. 
Once you hit it a third time, it will cancel the mod, "releasing" the mod.

You're key tester won't recognize the first tap until the next tap, anyhow.  And the second tap is only a "key down".

mnpqraven, 2018-11-04:
yeah in this case even when the tapping number is achieved the mod keys still isn't held down like expected, it just does a single tap

equalunique, 2018-11-05:
I may be experiencing the same problem. OSM is configured per the QMK Docs instructions, but tapping to hold the mod doesn't work 99% of the time. I have been using this layout for a day now and only once observed a mod held (I am assuming due to tapping), but only after some random button mashing.

Per my keymap here: https://github.com/equalunique/qmk_firmware/blob/master/keyboards/xd75/keymaps/equalunique-OrgoV2/keymap.c

drashna, 2018-11-06:
@mnpqraven what keyboard are you using? 

@equalunique those defines MUST be in the config.h file or they won't be picked up, at all. 
```c
// Settings for One Shot Mods (i.e, OSM()).
#define IGNORE_MOD_TAP_INTERRUPT
#define TAPPING_TERM 300
#define ONESHOT_TAP_TOGGLE 3  /* Tapping this number of times holds the key until tapped this number of times again. */
#define ONESHOT_TIMEOUT 3000  /* Time (in ms) before the one shot key is released */
```

equalunique, 2018-11-06:
@drashna Thank you, that took remedied my issue. Good luck, @mnpqraven !

drashna, 2018-11-06:
@equalunique awesome.

@mnpqraven Unfortunately, I think that may be an issue with the VUSB based boards.  I've seen others with issues with Tap Dance.  Both use software timers, so I think the issue is in that functionality.

And ... fixing that is way over my head.