qmk/qmk_firmware #4335: You need to clone the qmk_firmware repository outside the linux filesystem.

HMarxen, 2018-11-03:
Hi,
Msys2 prompted weird error codes so I installed the Ubuntu Windows app. Setting this up was super easy. However, when I try to run ./util/qmk_install.sh it prompts:
You need to clone the qmk_firmware repository outside the linux filesystem.
Otherwise the windows executables can't be run.
How do I do this? When I google the prompt, I just get to this github repo.
Can I even use the Ubuntu app for this?

drashna, 2018-11-03:
https://docs.qmk.fm/#/getting_started_build_tools?id=windows-10-deprecated

MSYS is the better method to use, to be honest, because of this (and some other odd issues). 

But basically, you need to download git for windows, install that, use that to clone qmk_firmware, and then run the rest from LxSS/WSL/Bash for Windows.

HMarxen, 2018-11-03:
THX! I had changed because I got this prompt for which I couldn't find anything helpful when I googled it.
 
   There are 55 members in group base-devel: :: Repository msys
    1) asciidoc 2) autoconf 3) autoconf2.13 4) autogen 5) automake-wrapper
    6) automake1.10 7) automake1.11 8) automake1.12 9) automake1.13
    7) automake1.14 11) automake1.15 12) automake1.6 13) automake1.7
    14) automake1.8 15) automake1.9 16) bison 17) diffstat 18) diffutils
    19) dos2unix 20) file 21) flex 22) gawk 23) gdb 24) gettext
    25) gettext-devel 26) gperf 27) grep 28) groff 29) help2man
    30)intltool 31) lemon 32) libtool 33) libunrar 34) libunrar-devel
    35) m4 36) make 37) man-db 38) pacman 39) pactoys-git 40) patch
    41) patchutils 42) perl 43) pkg-config 44) pkgfile 45) quilt 46) rcs
    47) scons 48) sed 49) swig 50) texinfo 51) texinfo-tex 52) ttyrec
    53) unrar 54) wget 55) xmlto

coarse, 2018-11-03:
If you still want to push through with using Ubuntu on Windows, you can access your hard drive in the `/mnt` directory.

I have all my git repositories stored in `C:/dev/`, so I access them on Ubuntu on Windows via `cd /mnt/c/dev/`.

HMarxen, 2018-11-03:
THX. I've got it all running. :)
But one strange thing:
The example from the instructions compiles just fine.

~/qmk_firmware$ make clueboard/66/rev3:default 

but when I try to compile my own keyset this appears:

~/qmk_firmware$ make xd75/keymaps/henning:default
QMK Firmware 0.6.164
make: *** No rule to make target 'xd75/keymaps/henning:default'. Stop.
|
|  QMK's make format recently changed to use folder locations and colons:
|     make project_folder:keymap[:target]
|  Examples:
|     make planck/rev4:default:dfu
|     make planck:default

This also compiles fine:
~/qmk_firmware$ make xd75:default
I hav tripple checked the folder path.

pevecyan, 2018-11-03:
Hi, 
try to compile with the following command:
`make xd75:henning`

HMarxen, 2018-11-03:
That prompts:
```
Compiling: keyboards/xd75/xd75.c                                                                   In file included from <command-line>:0:0:
./keyboards/xd75/keymaps/henning/config.h:21:10: error: #include expects "FILENAME" or <FILENAME>
 #include keymap_extras/keymap_german.h
          ^
 [ERRORS]
 |
 |
 |
tmk_core/rules.mk:356: recipe for target '.build/obj_xd75_henning/keyboards/xd75/xd75.o' failed
make[1]: *** [.build/obj_xd75_henning/keyboards/xd75/xd75.o] Error 1
Make finished with errors
Makefile:529: recipe for target 'xd75:henning' failed
make: *** [xd75:henning] Error 1
```

pevecyan, 2018-11-03:
Okay, that one is error in your keymap, try to replace `#include keymap_extras/keymap_german.h` with *#include keymap_german.h*.

HMarxen, 2018-11-03:
Thank you very much! I guess I rushed ahead without reading the next topic. If I may ask you about one last issue? I get a huge error message about rgb light. But apart from including RGB mode keys in my layout I didn't do anything with RGB.
```
Compiling: keyboards/xd75/keymaps/henning/keymap.c                                                  [OK]
Compiling: quantum/rgblight.c                                                                      quantum/rgblight.c:47:0: error: "MIN" redefined [-Werror]
 #define MIN(a,b) (((a)<(b))?(a):(b))
 ^
In file included from lib/lufa/LUFA/Drivers/USB/USB.h:382:0,
                 from tmk_core/protocol/usb_descriptor.h:47,
                 from tmk_core/common/report.h:83,
                 from quantum/keymap.h:32,
                 from quantum/keymap_extras/keymap_german.h:20,
                 from ./keyboards/xd75/keymaps/henning/config.h:21,
                 from <command-line>:0:
lib/lufa/LUFA/Drivers/USB/../../Common/Common.h:179:0: note: this is the location of the previous definition
     #define MIN(x, y)               (((x) < (y)) ? (x) : (y))
 ^
quantum/rgblight.c:48:0: error: "MAX" redefined [-Werror]
 #define MAX(a,b) (((a)>(b))?(a):(b))
 ^
In file included from lib/lufa/LUFA/Drivers/USB/USB.h:382:0,
                 from tmk_core/protocol/usb_descriptor.h:47,
                 from tmk_core/common/report.h:83,
                 from quantum/keymap.h:32,
                 from quantum/keymap_extras/keymap_german.h:20,
                 from ./keyboards/xd75/keymaps/henning/config.h:21,
                 from <command-line>:0:
lib/lufa/LUFA/Drivers/USB/../../Common/Common.h:165:0: note: this is the location of the previous definition
     #define MAX(x, y)               (((x) > (y)) ? (x) : (y))
 ^
cc1: all warnings being treated as errors
 [ERRORS]
 |
 |
 |
tmk_core/rules.mk:356: recipe for target '.build/obj_xd75_henning/quantum/rgblight.o' failed
make[1]: *** [.build/obj_xd75_henning/quantum/rgblight.o] Error 1
Makefile:529: recipe for target 'xd75:henning' failed
Make finished with errors
make: *** [xd75:henning] Error 1
```

pevecyan, 2018-11-03:
Huh, this seems to be a bit out of my scope, but for sure it would help us to resolve this issue if you provide your keymap files.

HMarxen, 2018-11-03:
I didn't really change anything from the default except the key layout.
```

#include "xd75.h"

// Layer shorthand
#define _QW 0
#define _FN 1

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {

 [_QW] = { /* QWERTY */
  { KC_ESC,	 KC_VOLD, KC_VOLU, KC_MUTE, KC_PSCR, DE_DQOT, DE_SLSH, DE_ASTR, DE_MINS, DE_AT,   DE_SS,   DE_HASH, DE_QUOT, KC_DEL,  KC_BSPC },
  { KC_TAB,  DE_Q,    DE_W,    DE_E,    DE_R,    DE_T,    DE_7,    DE_8,    DE_9,    DE_Z,    DE_U,    DE_I,    DE_O,    DE_P,    DE_UE   },
  { KC_CAPS, DE_A,    DE_S,    DE_D,    DE_F,    DE_G,    DE_4,    DE_5,    DE_6,    DE_H,    DE_J,    DE_K,    DE_L,    DE_OE,   DE_AE   },
  { KC_LSFT, DE_Y,    DE_X,    DE_C,    DE_V,    DE_B,    DE_1,    DE_2,    DE_3,    KC_N,    DE_M,    DE_COMM, DE_DOT,  KC_UP,	  MO(_FN) },
  { KC_LCTL, KC_LGUI, KC_LALT, DE_LESS, DE_ACUT, KC_ENT,  DE_0,    KC_PDOT, DE_PLUS, KC_SPC,  KC_SPC,  DE_ALGR, KC_LEFT, KC_DOWN, KC_RGHT },
 },

 
 [_FN] = { /* FUNCTION */
  { KC_CALC, KC_MPRV, KC_MNXT, KC_MPLY, KC_MSTP, _______, KC_F10,  KC_F11,  KC_F12,  _______, _______, _______, _______, _______, _______ },
  { _______, _______, _______, _______, _______, _______, KC_F7,   KC_F8,   KC_F9,   _______, _______, _______, _______, _______, _______ },
  { _______, _______, _______, _______, _______, _______, KC_F4,   KC_F5,   KC_F6,   _______, _______, _______, _______, _______, _______ },
  { _______, BL_STEP, _______, _______, RESET,   _______, KC_F1,   KC_F2,   KC_F3,   RGB_HUI, RGB_SAI, RGB_SPI, RGB_VAI, RGB_MOD, _______ },
  { BL_DEC,  BL_TOGG, BL_INC,  _______, _______, _______, _______, _______, _______, RGB_HUD, RGB_SAD, RGB_SPD, RGB_VAD, RGB_RMOD,RGB_TOG },	 
 }
};

const uint16_t PROGMEM fn_actions[] = {

};

const macro_t *action_get_macro(keyrecord_t *record, uint8_t id, uint8_t opt)
{
  // MACRODOWN only works in this function
      switch(id) {
        case 0:
          if (record->event.pressed) {
            register_code(KC_RSFT);
            #ifdef BACKLIGHT_ENABLE
              backlight_step();
            #endif
          } else {
            unregister_code(KC_RSFT);
          }
        break;
      }
    return MACRO_NONE;
};
```

coarse, 2018-11-03:
Replace line 47 and 48 of [`quantum/rgblight.c`](https://github.com/qmk/qmk_firmware/blob/master/quantum/rgblight.c#L47) with this:

```C
#ifndef MIN
  #define MIN(a,b) (((a)<(b))?(a):(b))
#endif
#ifndef MAX
  #define MAX(a,b) (((a)>(b))?(a):(b))
#endif
```

What's happening is that the macros `MIN` and `MAX` were being redefined in `rgblight.c`, but they were already defined in the LUFA USB driver.

HMarxen, 2018-11-04:
hmmm now I get a million error codes about every bit of the keymap and alot about usb. I had followed the instructions about the drivers. However, the board came with tmk and I had folowed those driver instructions first. Is there perhaps a conflict now?

drashna, 2018-11-05:
You need to have `#include "keymap_german.h"` in your keymap.c file, or it will complain about all of the `DE_*` keys

markstos, 2019-07-29:
There's been no follow-up from the post author for months. Can this be closed?

closed 2019-11-21 by drashna