qmk/qmk_firmware #4336: TMK Keyboard USB converted compatibility

yar05, 2018-11-03:
Hi, Is there any way to get this working with the  Hasu USB2USB converter ?

https://www.1upkeyboards.com/shop/controllers/usb-to-usb-converter/

drashna, 2018-11-03:
We already have support for it. :)

https://github.com/qmk/qmk_firmware/tree/master/keyboards/converter/usb_usb/hasu

yar05, 2018-11-03:
So I've tried to use the editor and that keymap to be compiled by importing below : 

{"keyboard":"converter/usb_usb/hasu","keymap":"converter/usb_usb/hasu_keymap_mine","layout":"KEYMAP","layers":[ [ "Esc", { "x": 1 }, "F1", "F2", "F3", "F4", { "x": 0.5 }, "F5", "F6", "F7", "F8", { "x": 0.5 }, "F9", "F10", "F11", "F12", { "x": 0.25 }, "PrtSc", "Scroll Lock", "Pause\nBreak" ], [ { "y": 0.5 }, "~\n`\n\n\n\n\n#", "!\n1", "\"\n2", "£\n3", "$\n4", "%\n5", "^\n6", "&\n7", "*\n8", "(\n9", ")\n0", "_\n-", "+\n=", { "w": 2 }, "Backspace", { "x": 0.25 }, "Insert", "Home", "PgUp" ], [ { "w": 1.5 }, "Tab", "Q", "W", "E", "R", "T", "Y", "U", "I", "O", "P", "{\n[", "}\n]", { "w": 1.5 }, "|\n\\", { "x": 0.25 }, "Delete", "End", "PgDn" ], [ { "w": 1.75 }, "Caps Lock", "A", "S", "D", "F", "G", "H", "J", "K", "L", ":\n;", "@\n'", { "w": 2.25 }, "Enter" ], [ { "w": 2.25 }, "Shift", "Z", "X", "C", "V", "B", "N", "M", "<\n,", ">\n.", "?\n/", { "w": 2.75 }, "Shift", { "x": 1.25 }, "↑" ], [ { "w": 1.25 }, "Ctrl", { "w": 1.25 }, "Win", { "w": 1.25 }, "Alt", { "a": 7, "w": 6.25 }, "", { "a": 4, "w": 1.25 }, "Alt", { "w": 1.25 }, "Win", { "w": 1.25 }, "Menu", { "w": 1.25 }, "Ctrl", { "x": 0.25 }, "←", "↓", "→" ] ]}

Do you know why is it complaining that ESC is unexpected ?
Found an unexpected keycode 'Esc' on layer 0 in keymap. Setting to KC_TRNS
I have a feeling its a syntax error ?

noroadsleft, 2018-11-03:
Looking through the QMK source, it looks like all the layouts configured for this converter are for fullsize keyboards; there isn't a macro for TKL.

I can do something about this (and clean up the extra layouts) but it'll take some time.

drashna, 2018-11-03:
@noroadsleft you're half right.  It's for the full ANSI scancode list, not a keyboard layout in specific.
It reads the scancode from the keyboard plugged into it, and then converdst that into the matrix, and then outputs based on the location/keymap. 

It's really quite neat.

noroadsleft, 2018-11-03:
@drashna Then why are there four macros that all have different keys? I must be missing something here.

drashna, 2018-11-03:
For ease of use, actually.  If you look at `KEYMAP_all`, it's formatted based on the HID Scan codes, and not an actual layout

yar05, 2018-11-03:
Guys, So I copied the keymap from xd87 and generated one for hasu USB2USB. 
Trying to flash it but I get this : 
Validating...
24874 bytes used (86.75%)
make[1]: *** [tmk_core/avr.mk:144: dfu] Error 1
Make finished with errors
make: *** [Makefile:544: converter/usb_usb:converter_usb_usb_hasu_keymap_mine:dfu] Error 1

noroadsleft, 2018-11-03:
I literally just realized that about a minute ago, and was about to comment saying as much.

drashna, 2018-11-03:
Don't run this with sudo, it can cause issues. 

Try running `make clean`, and then recompile. 

And you may want to do this, so you don't need to run sudo:
https://docs.qmk.fm/#/faq_build?id=linux-udev-rules

yar05, 2018-11-03:
tried make clean
then make converter/usb_usb:default:dfu

Checking file size of converter_usb_usb_hasu_default.hex                                            [OK]
 * The firmware size is fine - 25386/28672 (3286 bytes free)
dfu-programmer: no device present.
Error: Bootloader not found. Trying again in 5s.
Bootloader Version: 0x00 (0)
Validating...
25386 bytes used (88.54%)
make[1]: *** [tmk_core/avr.mk:144: dfu] Error 1
Make finished with errors
make: *** [Makefile:544: converter/usb_usb:default:dfu] Error 1

I've also ran this without sudo now with udev rules.
So this doesn't compile with default either

yar05, 2018-11-03:
https://www.reddit.com/r/MechanicalKeyboards/comments/6cryfj/qmk_on_hasus_usbtousb_converter/ is this any help?

drashna, 2018-11-03:
What's on the line that it mentions?

Also, it does generate the hex, correct?
If so, try the QMK Toolbox and see if that helps.

ment0s, 2018-11-03:
Hmm looks like i needed converter/usb_usb/hasu:default:dfu, after this it also gives the error however keyboard is functional.
FYI - Toolbox on linux loads but doesnt do anything after clicking  the button.
Toolbox on windows is just glitchy, it says it sees the device but then says no device avaliable when trying to do any action. 

T

closed 2019-11-21 by drashna