qmk/qmk_firmware #4344: Can we make a firmware for Poker?

F-TD5X, 2018-11-04:
I mean firmware for iKBC poker (poker ii, new poker, new poker ii, poker 3).

> new poker II's Processor is HT32F1654, ARM-CortexM3
> And I don't know other's processor.

mechmerlin, 2018-11-04:
There's a project out there already for this: https://github.com/pok3r-custom/qmk_pok3r

From hearsay, it works quite well, just a bit hard to setup properly.

F-TD5X, 2018-11-08:
@mechmerlin And what about emergency rescue method? I have googled some blog about repair their poker with broken firmware, but seems they are all failed and have to ship it back to iKBC?

mechmerlin, 2018-11-16:
There currently is no QMK official firmware for the poker, that link I provided is the best out there currently. I haven't looked at it in depth so I'm unsure about emergency rescue method. You could try filing an issue on their repo and see if they respond.

closed 2019-12-21 by stale[bot]

Dirtydog5, 2025-03-08:
> There currently is no QMK official firmware for the poker, that link I provided is the best out there currently. I haven't looked at it in depth so I'm unsure about emergency rescue method. You could try filing an issue on their repo and see if they respond.

Does it work for new poker II?
I am a fish.