qmk/qmk_firmware #4359: arm-none-eabi-gcc: error: rdimon.specs: No such file or directory

manuel-arguelles, 2018-11-05:
I'm getting this error when linking. Where should the rdimon.specs be? I can't find it on the repository nor my system, am I missing something?

manuel-arguelles, 2018-11-05:
Missing newlib

closed 2018-11-05 by manuel-arguelles

drashna, 2018-11-05:
@manuel-arguelles would you mind listing all the packages that need to be installed with the package manager? 

Alternatively, if you wanted to create a section for this: https://github.com/qmk/qmk_firmware/blob/master/util/linux_install.sh

manuel-arguelles, 2018-11-06:
@drashna I'm using slackware linux, the required packages are not in the official repository, I could add an entry in the linux_install.sh telling the use to install such packages (manually) from slackbuilds.org, would that be ok?

drashna, 2018-11-06:
Yeah, I think that should be okay.  

Or at least, open a PR, and we can get some feedback about that and then decide. :)