qmk/qmk_firmware #4387: XD60/XD64 v3.0

ff4500, 2018-11-08:
Formally requesting a firmware for the XD60/XD64 from [KBrepublic](https://kprepublic.com/products/xd60-xd64-3-0-pcb-custom-mechanical-keyboard-kit-underglow-rgb-gh60-60-programmable-mx-alps-stem-split-spacebar-type-c)/[Deskcandy](https://www.deskcandy.xyz/group-buy/xd60-v30?category=pcbs).  
(Who names these things?)

There are a few others in the repo currently, but the v3.0 supports split spacebars and a bunch of layouts.

edulpn, 2018-11-14:
I've successfully flashed the Rev 2.0 firmware, everything seems to be working fine, except for RGB commands. The Rev 2.0 PCB had only one row of LEDs, so the RGB commands work only for the central strip.

drashna, 2018-11-14:
Probably need to increase the number of RGB LEDs in the config file then.

ff4500, 2018-11-14:
Did it work with split spacebars? That was my hurdle with the current firmware.

I also couldn't adjust the backlighting, but in fairness, I probably had the commands setup weird. Didn't check RGB because I have it in a 5° case currently.

edulpn, 2018-11-14:
@drashna yeah, probably. I'm going to tinker with it later.
@ff4500 nah, I'm not a split space bar guy. I don't have backlighting LED's installed, so I'm not sure if it works. RGB underglow works for 8 LED's only, the strip in the center of the PCB (the one in common with the Rev 2.0 PCB).

ff4500, 2018-12-31:
Noticed that there was already a working PR in for this just now, so referencing it here for exposure (and split spacebars, obv): https://github.com/qmk/qmk_firmware/pull/4457

edulpn, 2019-01-02:
Yes, I have some work done, just need to find some time to update with the requested changes by the code reviewers.
And sorry, still no split spacebar :(

ff4500, 2019-01-02:
Progress is better than nothing though @edulpn . I'll worry about split spacebars once the rest of the map is there.

closed 2019-12-21 by stale[bot]