qmk/qmk_firmware #4391: The currently supported 68keys board is a duplicate of the mf68

zvecr, 2018-11-08:
The currently supported 68keys board is a duplicate of the mf68. According to the docs, the [site](https://68keys.io/guide/board/) details the process of getting the mf68 gerber files printed without any modifications.

Suggestion: merge keyboard folders together, adding the existing 68keys default keymap to mf68/keymaps/68keys

drashna, 2018-11-09:
It may be, but it's different people doing it.  

I'd feel better about leaving them as separate boards.

drashna, 2018-11-13:
Because in this case, it may not be the same PCB.  

And in that case, we do have something like tht already with the planck, contra, fractal, and the like.

zvecr, 2018-11-13:
From the site i linked, 
The github link points to a repo with the following commit:
https://github.com/sbstjn/tmk_keyboard/commit/ffb1d7a5250d9d6219d00902885b532983ca0e1b
Add symlink to mf68 PCB

and an image from that page:
[image]

I have to say, Its a shame to hear that this awesome project advocates for code duplication.

drashna, 2018-11-14:
Ah, okay, it is the same PCB then.  In that case, yeah, don't need multiple boards.

closed 2018-11-16 by drashna

zvecr, 2018-11-16:
Cheers @drashna. Thanks for hearing me out on this one.

PR #4435 (2018-11-16): Unify 68keys with mf68
Merge keyboard folders together, adding the existing 68keys default keymap to mf68/keymaps/68keys

Fixes #4391

Let me know if any other changes are required.